prusa3d/PrusaSlicer #6401: [FR] Start Simulation for Duet

bttguy, 2021-04-21:
There is a final PR for this FR, I am just would like to aks what is the plan with it in regard to its implementation.
Start Simulation for Duet #5157
Sorry for opening an issue about it, but the PR get no response for more than a month now.
Thank you for the awesome slicer!

bttguy, 2021-06-22:
@rtyr Hi! Do you have any update or plan regarding this question? Thank you in advance! :)

bubnikv, 2021-06-22:
Thanks for your contribution. We are usually looking into feature pull requests after we are done with our own plan, that is after the public alpha.

What is the "simulation" for?

bttguy, 2021-06-22:
@bubnikv Thank you for the response and the clarification! This is from the duet wiki: "The ‘Simulate File’ option runs the file without heating any heaters or moving any axis. The firmware works out how long moves take and thus can determine an accurate estimate of the print time. When started, the screen switches to the ‘Status’ screen to show the simulation progress. Simulations run much faster than real prints, typically taking a couple of minutes per megabyte of G-Code. Simulations can be paused and cancelled from this screen."

So this would be a convinience feature to allow not just to start the print after upload, but to start simulate automatically if we want.

lukasmatena, 2024-02-18:
Merged long ago. Closing.

closed 2024-02-18 by lukasmatena