prusa3d/PrusaSlicer #6403: Printer qualities not matching your expectations

kingzeus123, 2021-04-21:
### Version
Mk3s+

The printer regardless of quality seems to be worse the better I place the quality in the slicer.

rtyr, 2021-04-21:
Sorry, I don't understand what are you trying to say. Is there any particular issue with the slicer?

kingzeus123, 2021-04-21:
The printer itself Is not printing at the quality it should

kingzeus123, 2021-04-21:
The prints are coming out bad

murk-sy, 2021-04-21:
What I assume is the issue that using better settings with finer layer heights decreases print quality, but that's not really a slicer problem.

Post photos of the parts printed and provide the 3d model/3mf file, otherwise nobody will be able to help you out

rtyr, 2021-04-21:
If you have some issues with the printer, please contact our tech-support via live chat here https://shop.prusa3d.com/ or by e-mail.

Closing.

closed 2021-04-21 by rtyr

kingzeus123, 2021-04-21:
Sent pictures the prints are getting worse on the finer print than the
regular ones.... the ring was on a regular setting which was better than
the ring in a better setting.... also at temp the nozzle is oozing
filiment.... I lowered the temp and even at 200 it seems to ooze filiment
making prints not stick fully on bed because it is going over the filiment
that oozes out