prusa3d/PrusaSlicer #6406: Mirror Inverts Exported Normals

deckar01, 2021-04-21:
### Version

2.3.1+x64

### Operating system type + version

MacOS 11.1

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK2.5S

### Behavior

Mirroring an object inverts the normals when exported as an STL.

1. Add an STL to the build plate
2. Right click on the object
3. Select "Mirror" > "Along (any) axis"
4. Right click on the object
5. Select "Export as STL"
6. Open the STL with a tool that indicates face normals

The normals should face outwards, but that are inverted inwards.

[image]

#### Project File (.3MF) where problem occurs

[Mirror Inverted Normal.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6352258/Mirror.Inverted.Normal.3mf.zip)

deckar01, 2021-04-22:
https://github.com/prusa3d/PrusaSlicer/blob/c1179fc2c7f9048a315a38d390e55d2052b1213d/src/libslic3r/TriangleMesh.cpp#L358-L360

The mirror operation is applied as a general transformation matrix that gets applied when the mesh is exported, but the normals are only being repaired when the model is left handed. I believe any transformation with a negative determinant needs to be repaired.

```cpp
    bool inverted = t.matrix().block(0, 0, 3, 3).determinant() < 0.;
    if (inverted)
        this->repair(false);
    if (fix_left_handed && inverted) {
        // Left handed transformation is being applied. It is a good idea to flip the faces and their normals.
```

lukasmatena, 2021-04-26:
Should be fixed with 978b359. @deckar01 , please review and test it (if you want to). Thanks.
Closing.

closed 2021-04-26 by lukasmatena

deckar01, 2021-04-26:
Unqualified to review, but I will absorb as much as I can. I was able to confirm that your change resolves the problem, and I did not hit any regressions when testing other transformations. Thank you.