prusa3d/PrusaSlicer #6407: Sliced output missing top layer of text from model

teknogeek, 2021-04-21:
### Version
PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Windows 10 Pro Build 19042, 64-bit

### 3D printer brand / version + firmware version (if known)
- Original Prusa MINI (not MINI+), unmodified
- OEM firmware, version 4.2.0-final+2067

### Behavior
When slicing a model with small text on top, PrusaSlicer has trouble slicing the text on top and even ignores it entirely in some cases. Sometimes it is possible to get better results by changing the lower height to something very small (under 0.10mm), but it doesn't fix it every time.

Here are two models that I am trying to print, which both exhibit this problem:

1. https://www.thingiverse.com/thing:1091544
2. https://www.thingiverse.com/thing:3552299

In PrusaSlicer, this is what the models look like:
[image]

But when you slice them at 0.20mm and 15%, it comes out looking like this:

[image]

Even if I drop the layer height to the lowest ULTRADETAIL setting of 0.05mm, it still comes out like this:

[image]

As you can see, one of the models _almost_ comes out okay, but the other still is missing most of the text and a small portion now appears. Additionally, where the text does show, it has strange gaps where it should be solid:

[image]

#### Project File (.3MF) where problem occurs

3MF (in a zip): [SmallText.zip](https://github.com/prusa3d/PrusaSlicer/files/6352299/SmallText.zip)

rtyr, 2021-04-21:
The model is not suitable for 0.4mm nozzle. The text is too tiny to be printed with 0.4mm extrusion width. You can try "detect thin walls" parameter, but I would not expect good results considering that even 0.2mm extrusion width is too much.

rtyr, 2021-04-26:
Answered. No response. Closing.

closed 2021-04-26 by rtyr