prusa3d/PrusaSlicer #6408: How to add a new profile for my Creality printer ?

kot0005, 2021-04-21:
Anyone have a profile file they can share ? i was setting up a profile for my CR 10s pro but receive this error :

G-code export to C:\Users\user\AppData\Local\Temp\.28212.gcode failed due to invalid custom G-code sections:
end_gcode
Parsing error at line 8: Not a variable name
G1 X0 Y{machine_depth} ;Present print
        ^
Please inspect the file C:\Users\user\AppData\Local\Temp\.28212.gcode.tmp for error messages enclosed between
        !!!!! Failed to process the custom G-code template ...
and
        !!!!! End of an error report for the custom G-code template ...
for all macro processing errors.

heres my start and end g codes in printer settings. i use these in Cura.

Start:

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G90; Switch to Absolute XYZ
G28 ;Home
G29 ;Auto Bed leveling
M420 S1 Z2 ;Enable ABL using saved Mesh and Fade Height

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

End:

G91 ;Relative positioning
G1 E-20 F2400 ; retract 20mm at 2400mm/minute
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

rtyr, 2021-04-21:
machine_depth is Cura specific placeholder.

We have CR 10S Pro profile bundled in PrusaSlicer 2.3.1, so you may just use that one.

kot0005, 2021-04-21:
> machine_depth is Cura specific placeholder.
> 
> We have CR 10S Pro profile bundled in PrusaSlicer 2.3.1, so you may just use that one.
yeah i know 2.3.1 has it but i couldnt find one in mine :(

i can only see ender 3 profile.

[image]
 i uninstalled and re installed as well.

kot0005, 2021-04-22:
nvm i manually imported them from github and now it works

closed 2021-04-22 by kot0005