prusa3d/PrusaSlicer #6409: Overhang Perimeter and Bridging Infill

OopAck, 2021-04-21:
### Version
2.3.1

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Overhang Perimeter and Bridging Infill is being generated and increasing or change the size of the print.
And also destroying the specific face that it appears on

#### Project File (.3MF) where problem occurs
It is appearing on any support that is being generated.
 
Requesting away to remove it from the print afterwards or not to have it print at all.
[image]

rtyr, 2021-04-21:
Implemented by https://github.com/prusa3d/PrusaSlicer/commit/ceea9de8b8b1d4b35bdd1180051033680c405ba1

See https://github.com/prusa3d/PrusaSlicer/issues/5793#issuecomment-802878625

Closing.

closed 2021-04-21 by rtyr

OopAck, 2021-04-21:
I have Slicer 2.3.1 and not seeing Think bridges.

rtyr, 2021-04-21:
It is implemented on master, it will be included in 2.4.0 releases.

OopAck, 2021-04-21:
Is there away I might be able to get ahold of the master to test it out? Barry Sent from Mail for Windows 10 From: rtyrSent: Wednesday, April 21, 2021 3:06 PMTo: prusa3d/PrusaSlicerCc: OopAck; AuthorSubject: Re: [prusa3d/PrusaSlicer] Overhang Perimeter and Bridging Infill (#6409) It is implemented on master, it will be included in 2.4.0 releases.