prusa3d/PrusaSlicer #6411: Extra filament being extruded at points on first layer.

huntervaneck, 2021-04-21:
Version:   2.3.1+win64
Build:     PrusaSlicer-2.3.1+win64-202104161404

Windows 10 (build 19042), 64-bit edition

SOVOL SV01
Marlin 2.0.5.3

**Behavior**
When slicing a model with edges that come to points, PrusaSlicer adds extra bits of filament to the end of the point, but only on the first layer. This can be seen in the slicer and on the finished print.

[image]
[image]

rtyr, 2021-04-21:
Disable "detect thin walls".

huntervaneck, 2021-04-22:
Ah, you're awesome. That fixed the problem, thank you!

closed 2021-04-22 by rtyr