prusa3d/PrusaSlicer #6412: PrusaSlicer 2.3.1 not responding

rmilyard, 2021-04-21:
### Version
2.3.1

Windows 10 Pro x64

Installed it today and now whenever try load just get not responding.  
[image]

rmilyard, 2021-04-22:
Uninstalled all and reinstalled now.  Now just getting this:

[image]

nenovidenovic, 2021-04-25:
I have this issue as well on my new laptop. Works fine on my old laptop.

cpeuschel, 2021-06-13:
Same for me. If the Slicer starts on second or third screen, the programm is not responding. 
But if the Slicer ist started on the primary screen every works fine.

neophyl, 2021-06-13:
Do a search for nahimic.  There are already several of these reported and its usually down the having a nahimic driver installed.

bubnikv, 2021-09-25:
May be similar to https://github.com/prusa3d/PrusaSlicer/issues/2939

It may also be an OpenGL driver issue. Often dual graphic cards cause issues, disabling the more powerful graphics card may help.

lukasmatena, 2023-02-22:
Is this still happening with 2.6.0-alpha4?

TerPuc, 2025-04-28:
Dear @rmilyard, we are closing this issue due to inactivity. If the problem persists, please open a new issue. Thank you.

closed 2025-04-28 by TerPuc