prusa3d/PrusaSlicer #6413: Dual Extrusion

hugomsmarreiros, 2021-04-21:
Version

2.3.1

Beebee creative B2X300 Marlin 1.1.8

When using dual extrusion with wipe tower the second extruder is not heated for the first layer.

Also, could you make possible to have stand by temp when using the wipe tower?

Thanks in advance.

rtyr, 2021-04-22:
Please provide your 3MF project file.

hugomsmarreiros, 2021-04-22:
Hi.

in attach i sent you the 3mf and the config bundle.

also i forgot to mention if you select prime all printing extruders, it tries to print something out of bed.

[Dual Extrusion.zip](https://github.com/prusa3d/PrusaSlicer/files/6361222/Dual.Extrusion.zip)

rtyr, 2021-04-23:
> When using dual extrusion with wipe tower the second extruder is not heated for the first layer.

Remove all M104/M109 if you want PrusaSlicer to generate such commands automatically, otherwise you have to specify it manually in the start g-code.

[image]

> Also, could you make possible to have stand by temp when using the wipe tower?

You should be able to achieve that using tool change g-code script.

>  if you select prime all printing extruders, it tries to print something out of bed.

It seems to be related to your custom bed shape.

hugomsmarreiros, 2021-04-23:
gonna try the first part.

why not have an option like in ooze shield to have temperature variation?

rtyr, 2021-04-26:
> why not have an option like in ooze shield to have temperature variation?

It is currently not supported with wipe tower enabled.

> if you select prime all printing extruders, it tries to print something out of bed

Fixed with https://github.com/prusa3d/PrusaSlicer/commit/5783cc62fb02c7433b119026ffcbd574d853fd6d

> gonna try the first part.

Is it OK, can we close this issue?

hugomsmarreiros, 2021-04-28:
hi,

sorry for the late response...
yes you can close it, but i think it would be nice to add temp variation on the wipe tower to.

thanks.

closed 2021-04-28 by hugomsmarreiros