prusa3d/PrusaSlicer #6419: Support Overhang Problem

MarcAhoy, 2021-04-22:
_2.3.1+win64

windows 10 home 

modified anycubic i3 mega with Klipper firmware

In the settings, my support contact distance is 0.2mm, but when the supports are generated the 0.2mm clearance is from a little extrusion trying to support the hole. So the clearance from the flat surface of the overhang to the part of the support that touches it is 0.6mm not 0.2mm. This leads to having really bad overhangs that are barely supported because the support interface is so far away from the actual overhang, not the hole. 

[3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6360850/3mf.zip)
[image]
[image]
[image]
[image]

rtyr, 2021-04-22:
> that touches it is 0.6mm not 0.2mm
It is related to thick bridge flow used for bottom most layer of the supported object. It is correct (0.2mm contact distance + 0.4mm round extrusion). See #102 and other linked issues for more information.

We have recently added a new parameter, that will allow to optionally disable standard thick bridges. See https://github.com/prusa3d/PrusaSlicer/issues/102#issuecomment-802206617. This option will be available in PS2.4.0.

 `To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio.`

You can reduce your contact Z distance value to improve the result.

Closing.

closed 2021-04-22 by rtyr

MarcAhoy, 2021-04-23:
Oh, so the bridge layer is thicker than the 0.2mm layer height