prusa3d/PrusaSlicer #6421: Building on Linux: assert "uiIndex < m_nCount" failed in Item()

gringer, 2021-04-23:
### Version

version_2.4.0-alpha0-194-g657d19482

### Operating system type + version

Debian frankenLinux: sid / bullseye / buster

    $ uname -a
    Linux elegans 5.6.0-1-amd64 #1 SMP Debian 5.6.7-1 (2020-04-29) x86_64 GNU/Linux

### 3D printer brand / version + firmware version (if known)

N/A. I'm only testing out the UI.

### Behavior

Directly after showing the splash screen (i.e. after running `prusa-slicer` from the command line), this happens:

ASSERT INFO:
../include/wx/dynarray.h(838): assert "uiIndex < m_nCount" failed in Item().

BACKTRACE:
[1] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[2] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[3] wxEvtHandler::TryHereOnly(wxEvent&)
[4] wxEvtHandler::ProcessEventLocally(wxEvent&)
[5] wxEvtHandler::ProcessEvent(wxEvent&)
[6] wxNotebookBase::SendPageChangedEvent(int, int)
[7] g_closure_invoke
[8] g_signal_emit_valist
[9] g_signal_emit
[10] wxNotebook::InsertPage(unsigned long, wxWindow*, wxString const&, bool, int)
[11] wxEntry(int&, wchar_t**)
[12] __libc_start_main

I have the option of *Stop* or *Continue*. Clicking *Continue* leads directly to a segmentation fault.

* _Expected Results_
  * Seeing a user interface

* _Actual Results_
  * Kicked out of the application
[image]

gringer, 2021-04-23:
I tried compiling using `cmake -DSLIC3R_STATIC=1 ..`, and got the following pre-compilation error:

```
CMake Error at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  TBB library cannot be found.  Consider set TBBROOT environment variable.
  (missing: TBB_LIBRARIES) (found version "2020.3")
```

gringer, 2021-04-24:
I've been able to get it to compile with static libraries using the *deps* directory:

```
cd deps/
mkdir build; cd build
cmake .. -DDESTDIR=/home/${USER}/install/PS
make
cd ..
mkdir build; cd build
cmake -DSLIC3R_STATIC=1 -DCMAKE_PREFIX_PATH=/home/grinja/install/PS/usr/local ..
make
```

Kristaba, 2021-04-27:
Same issue on Arch Linux, when building `master` branch.
I will try building the `version_2.4.0-alpha0` tag just in case, and build with static libraries if it still fails as proposed by @gringer .

```
$ uname -a
Linux archlinux 5.11.14-arch1-1 #1 SMP PREEMPT Wed, 14 Apr 2021 12:06:34 +0000 x86_64 GNU/Linux
```
I am not familiar with PrusaSlicer code nor wxWidgets, but if someone has a few pointers or a patch to test, I can probably manage to find some time.

Kristaba, 2021-04-27:
I missed the discussion on the [Arch AUR thread](https://aur.archlinux.org/packages/prusa-slicer-git/#comment-799631), which links back to another issue here. As it refers to the same problem, either this one or https://github.com/prusa3d/PrusaSlicer/issues/6270 should be marked as duplicated.

gringer, 2021-05-06:
This issue is a duplicate of #6270

closed 2021-05-06 by gringer