prusa3d/PrusaSlicer #6423: [Feature Request] Per Layer Extruder/Tool Change

sarf2k4, 2021-04-23:
### Version 2.3.1+win64
### Windows 10 64-bit
### Prusa MK3s MMU2s
Would like to request a new feature for per layer extruder assignment that are much easier to inject rather than manually doing it. I wanted to have somewhat an alternating colour 3d printed object shown below
[image]
The image above is what I want to achieve, however, with the current setup, per layer extruder assignment is very tedious where, I have to scroll every layer and do multiple clicks in order to achieve similar to the image above.

My proposal is, to add at least 2 more textboxes:

1. Extruder Sequence - This will instruct the slicer to automatically do a change of extruder manually assigned by the user, in the case of the image above, it is 1,2,3,4,5 as its sequence when going up. The user will input extruder number with comma as separator for example "1,2" for only extruder 1 and 2, or "1,3,5" for extruder 1,3 and 5, and so on depending on the user's creativity and desire
2. Tool/Extruder Change On Every: - This behavior should be similar to "combine infill every:" option, for example the print layer height is 0.2mm, setting this option to 1, would change to the other extruder/tool upon layer change. If "3", it will change on every third layer

Below is my 3mf file for inspection
[DH010.zip](https://github.com/prusa3d/PrusaSlicer/files/6365539/DH010.zip)

rtyr, 2021-04-23:
This feature is already implemented (since 2.2.0-alpha1).

[image]

Closing.

closed 2021-04-23 by rtyr

JohnOCFII, 2021-04-24:
The User Interface visibility of recent features (those in physical printer, and this example here behind the "Gear" icon) could certainly be improved if there is a UI refresh in the future.