prusa3d/PrusaSlicer #6424: Object Fails to Slice

thejobbitt, 2021-04-23:
### Version
2.3.1

### Operating system type + version
Windows 10 Pro 10.0.19043 Build 19043

### 3D printer brand / version + firmware version (if known)
Stock Prusa i3 mk3s
3.9.3-3556

### Behavior
Slicer hangs while trying to slice this object. 
[image]

#### Project File (.3MF) where problem occurs
[Primer_Flipping_Trays.zip](https://github.com/prusa3d/PrusaSlicer/files/6365755/Primer_Flipping_Trays.zip)

combolek, 2021-04-23:
It does eventually complete, but it takes a long time:
[image]

thejobbitt, 2021-04-23:
You are correct, took about a half hour. Thanks for the help.

closed 2021-04-23 by thejobbitt