prusa3d/PrusaSlicer #6426: Request raft function

viper0078, 2021-04-24:
### Version
2.3.1
### Operating system type + version
Win10 64bit

### Behavior
This is a request regarding Raft.
I mainly use Simplify3D.
I tried to use Cura, but it had a lot of parameters and it was difficult.
I tried to use PrusaSlicer and liked it because it has very good features.
However, compared to other software, the Raft function is unique and difficult to use.

1. The layer above the Raft is shown as a thick blue line. Is this a bridge?
　　Why does it become a bridge when it is supposed to have support (Raft)?
　　Is there anything I can do to prevent it from becoming a bridge?

Is the gap above the raft the same parameter as the gap above the support?
　　I would like this setting to be separated.
　　It seems that there are other parameters that are the same for the raft and the support.
　　Please separate them as well.

3. I would like to be able to change the thickness of the raft layers.
　　The surface of sheets such as BuildTak can become rough and uneven with use.
　　In this case, increasing the thickness of the first layer of raft will absorb the unevenness and make the BuildTak last longer.
　　BuildTak can be used for a long time. 

4.I would like to be able to control the speed of the layer that is printed directly above the raft.
　　By printing at a low speed and with an appropriate gap, we can adjust the fusing and peeling ease.

I would like to have separate support and raft parameters like other software.
I think this is to simplify the software, but I feel it is a disadvantage for the user.

rtyr, 2021-04-24:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3425.

We recently implemented raft enhancements, see https://github.com/prusa3d/PrusaSlicer/pull/6003. It will be included in 2.4.0.

Closing.

closed 2021-04-24 by rtyr

rtyr, 2021-04-24:
Speed setting for 1st layer of the raft is also planned.