prusa3d/PrusaSlicer #6427: Switch inherited profile

freiser77, 2021-04-24:
### Behavior
Hello,

I use the MK3S with 0.4mm and 0.6mm nozzle. When I build a custom printing profile for the 0.4mm nozzle, I would often like to migrate it to the 0.6mm nozzle. The customized properties should be migrated, the default ones set to the default for the 0.6mm nozzle.

I wish there was something like a possibility to switch the inherits property to some other profile, like "0.30mm@MK3" to "0.30mm QUALITY@0.6mm MK3"

Greetings

_Is this a new feature request?_: yes

rtyr, 2021-04-28:
I am not sure I understand what exactly do you want (and how to implement it).

What about https://github.com/prusa3d/PrusaSlicer/issues/5384, wouldn't it be better solution? (profile comparison with transfer option).

freiser77, 2021-05-02:
It should be possible to switch the inherits field (shown on "Print Settings" -> Dependencies).
Steps needed:
1. Create diff to old inherits profile
2. Switch inherits field
3. Apply diff to new inherited profile.

A simple comparison wouldn't be better, because I would have to have a look at each and every setting, which leads inevitably to overlooking some changes.

rtyr, 2021-05-05:
There is already an issue related to this https://github.com/prusa3d/PrusaSlicer/issues/1108. I am closing this one, it will be linked to 1108.

closed 2021-05-05 by rtyr