prusa3d/PrusaSlicer #6430: Sliced distances between the walls too big

SphaeroX, 2021-04-24:
### Version
Version: 2.3.0+linux-x64
Build: PrusaSlicer-2.3.0+linux-x64-202101111322

### Operating system type + version
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 5.4.0-72-generic x86_64

### 3D printer brand / version + firmware version (if known)
Klipper

### Behavior
Hey Guys,

PrusaSlicer makes the distances between the walls too big and I can't find a setting that prevents this. Neither XY size compensation nor experiments with the extrusion width help. Does anyone have any advice for me?

I have already scaled the model to different sizes...

[image]

[image]

My Config: 

https://nopaste.chaoz-irc.net/view/74f2d093

uses "0.4mm" Profile with "Janbex PLA Grau"

Link to Model: https://www.thingiverse.com/thing:4821342/files

SphaeroX, 2021-04-25:
Well same troubles at Cura, anyway... is there an opton to make the dimensions more true to the original on the outside?

[image]

ryankesavan, 2021-04-26:
Is this the issue?
[https://www.thingiverse.com/thing:4821342](url)
> UPDATE 19Apr21: I have updated all .STLs to V2.0, which should make printing even easier (V2.0 adds 0.05 mm clearance to the cage only).

ryankesavan, 2021-04-27:
try changing the horizontal expansion to 0.05

SphaeroX, 2021-04-27:
> try changing the horizontal expansion to 0.05
I could reduce it with the XY size compensation

> Is this the issue?
It looks like with the new one it comes out better

I think the ticket can be closed, because it was a general problem and not directly from the slicer.

closed 2021-04-27 by SphaeroX