prusa3d/PrusaSlicer #6433: libGLX issue PS 2.3.1 under Linux

eiffel31, 2021-04-25:
PrusaSlicer 2.3.1

Linux Debian system

$ ./PrusaSlicer-2.3.1+linux-x64-202104161339.AppImage 
/tmp/.mount_PrusaSbnK2F8/usr/bin/bin/prusa-slicer: error while loading shared libraries: libGLX.so.0: cannot open shared object file: No such file or directory

Previous version 2.3.0 (PrusaSlicer-2.3.0+linux-x64-202101111322.AppImage) is working fine on this system.

OpenGL works fine on this system. There is no GLX lib installed.

MS3FGX, 2021-04-27:
I'm seeing the same error on Linux Mint Debian Edition (LMDE) 3, which is based on Debian 9.3. It looks like this file is part of the NVidia GPU drivers (at least on Debian 9), and I'm a bit skeptical of installing them when I don't have Nvidia hardware on this machine.

openelearning, 2021-05-16:
Same problem here with debian 9.13 with PrusaSlicer 2.3.1, solved using PrusaSlicer 2.3.0.

MS3FGX, 2021-07-10:
Seeing same issue on latest PrusaSlicer 2.3.2.

bubnikv, 2021-07-12:
I am not an expert on modern Linux distros, but does this help?

https://packages.ubuntu.com/search?searchon=contents&keywords=libGLX.so.0&mode=exactfilename&suite=hirsute&arch=any

eiffel31, 2021-10-03:
The problem remains with release 2.3.2, 2.3.3 and 2.4.0-alpha3, so I updated the title acordingly.

I'm still forced to stay on 2.3.0 release, which is the most recent release to be able to run on my linux system.

Is this GLX problem related to the building process of the AppImage? Is there a difficulty to go back to the previous building configuration?

bubnikv, 2021-10-04:
@eiffel31 The problem with Linux is that there is no Linux / desktop platform, meaning that there is no guarantee of system libraries and their particular versions being available on particular Linux installation. We and the appimage team heuristically tries to put together a list of libraries and their versions that are "likely" available at most linux installations. If those libraries are not available, then we may try to bundle the particular missing library with our appimage. This works as long as the bundled library is not a system one and / or does not have too many other dependencies outside the AppImage. It is a difficult topic.

You may try your luck to install PrusaSlicer from flatpack on your system.

You may also consider updating your system to a newer version.

sleemanj, 2021-10-08:
I assume by "flatpack" you mean the the tarball?  That is not going to work any different.

2.3.0 tarball works fine, the prusa-slicer binary doesn't link to libGLX so that is OK

    PrusaSlicer-2.3.0/bin$ ldd prusa-slicer | grep libGLX | wc -l
    0
    
2.3.1 and greater tarball does not work, the prusa-slicer binary does link to libGLX , which doesn't exist for the poster's above (and me)

    PrusaSlicer-2.3.1/bin$ ldd prusa-slicer | grep libGLX | wc -l
    1
    
I expect this is due to the OpenGL changes which were noted in the changelog of 2.3.1-rc1

libGLX is provided by the following in Ubuntu releases...

  * In Ubuntu focal (20.04) and later by `libglx-dev`
  * In Ubuntu bionic (18.04) by `libglvnd-dev`
  * It is not available in older releases

The [libglvnd](https://gitlab.freedesktop.org/glvnd/libglvnd) libraries, including libGLX compile just fine on the old Ubuntu 14.04, and PrusaSlicer 2.3.1 and above will actually startup, but they will not be functional as the GL canvases are not present unfortunately, this will be because the X.Org server in Ubuntu 14.04 is prior to version 1.19 which is when it started to support libglnvd.

s a result, if your operating system release (eg Ubuntu 14.04) runs a version of X.Org older than 1.19 your only practical recourse (apart from manually building X.Org as well) is to upgrade the operating system.  If you are on an operating system with X.Org 1.19 or newer, you may be able to find one of those packages `libglx-dev` or `libglvnd-dev` in your package manager, or manually compile `libglnvd` from the link above (it compiled cleanly on my old 14.04 as I say, not usefully, but cleanly).

Note that if somebody does want to experiment and manually compile and install libglnvd on a system that doesn't have a supporting X.Org, it will stop PrusaSlicer 2.3.0 and older from working too, so make sure you use stow or similar if you want to experiment so you can destow it again :-)

closed 2026-09-29 by prusaslicer-bot[bot]