prusa3d/PrusaSlicer #6436: Cannot add SLA materials

smidgleypivot, 2021-04-25:
### Version
2.3.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
SL1 

### Behavior
In PrusaSlicer go through the configuration wizard add multiple new resin SLA Materials the wizard finishes, but new materials do not appear in the drop down box when you try to select the new materials.
Tried reinstalling several times, to no avail. 

It is a common bug:
https://forum.prusaprinters.org/forum/prusaslicer/problem-configuring-with-different-filaments/

smidgleypivot, 2021-04-25:
Sorry, I attached the wrong link:
Please see the correct forum link
https://forum.prusaprinters.org/forum/prusaslicer/add-view-material/

rtyr, 2021-04-25:
Some SLA material profiles are available only for certain layer heights. For example "Prusa Blue Flexible" is currently available only for "0.05mm Normal" print setting.

rtyr, 2021-04-26:
Can we close this issue?

smidgleypivot, 2021-04-26:
I'm not sure that was the issue to be honest. I think on my system I can only initially add materials for the first time during the wizard. The second time you run the wizard the materials aren't updated. I was able to add additional materials by removing my user data from AppData and re-running the wizard.

rtyr, 2021-04-26:
Is it solved then? If not, can you reproduce it?

Sepitus, 2021-05-12:
i am able to add my own material in 2.3.1 (Elegoo ABS-Like Grey)

i do this by saving a present already there which comes up with a name box, i edit this name as needed then continue editing the new one as needed and then save

rtyr, 2021-07-23:
> The second time you run the wizard the materials aren't updated.

Fixed on master. Closing.

closed 2021-07-23 by rtyr