prusa3d/PrusaSlicer #6437: 2.3.1 weirdly shaped bed for Ender 3 v2 standard profile

sn0opy, 2021-04-25:
### Version
2.3.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 v2

### Behavior
I updated to 2.3.1 and wondered about certain field in my custom printer profil not matching the default anymore. Some of them were fine due to spelling fixes, but there's one that's somewhat weird. Why's the bed set to 210x220 with a negative origin offset? And more importantly: where's that change coming from? I was looking through the creality.ini but couldn't find a change between 2.3.0 and 2.3.1 that may result in this behaviour. 

[image]

rtyr, 2021-04-25:
@pmjdebruijn Could you check it please.

pmjdebruijn, 2021-04-26:
The original Ender-3 had a motion range of 235x235 (even though advertised as 220x220), which leaves some wiggle room for the prime line for example.

Sadly, Creality, limited the motion range to 220x220 on the Ender-3 V2, which seems like a weird move on their part, as it makes for a worse printer.

But we cut off 5mm off each side, to account for the priming line on the left. The right is slightly reduced as well so the center of the bed should remain the center.

Relevant commit:
https://github.com/prusa3d/PrusaSlicer/commit/138501338cd0aa039cb34e7ec4339ef954c91be7

rtyr, 2021-04-27:
@pmjdebruijn I am not sure about this. It means that an user can use only 210mm on X asis.

pmjdebruijn, 2021-04-27:
@rtyr: it does, otherwise there's the risk of a user hitting the prime line, since PrusaSlicer doesn't visualize the prime line on the print bed (in neither the 3D editor view or the Preview).

So this is a pick your poison situation, either we guard against hitting the prime line, or we offer the full bed. Can't have both. Given most people rarely use their full bed size, I've opted to default to "safety" on printers where Creality does not offer a large addressable range compared to their advertised bed size.

rtyr, 2021-04-27:
> PrusaSlicer doesn't visualize the prime line on the print bed (in neither the 3D editor view or the Preview).

It will be visualized in 2.4.0.

I understand your reasons, but I think that reducing advertised build volume by 10mm just because of intro line is not worth it. I guess we can print intro line closer to the edge (see Cura start gcode). It is also possible to remove it completely (we have skirt), but I think it is better to use intro line.

Cura (Ender-3):
```
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
```

pmjdebruijn, 2021-04-27:
The thing is, I don't want different behavior between different printers, which will devolve into a maintenance issue quite quickly.

I'd suggest leaving things as they are for 2.3.x, and changing this once 2.4.x is closer to an actual release.

pmjdebruijn, 2021-04-27:
BTW, will the prime line be visualized in the 3D editor view as well?

rtyr, 2021-04-27:
OK

Only g-code preview.

Closing.

closed 2021-04-27 by rtyr