prusa3d/PrusaSlicer #6438: No longer able to add custom gcode after upgrade.

galluccib, 2021-04-26:
### Version
PrusaSlicer 2.3.1

PrusaSlicer 2.3.1 +Win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
Prior to upgrade of 2.3.1 I was able to right click and add custom gcode. After the upgrade when I right click it just keeps bring the bottom to that point.

Pavulon87, 2021-04-26:
Do you have sequential printing enabled? This disables adding custom g-code.

galluccib, 2021-04-26:
Odd never ran into that before... But yes that was the issue.. THANKS!!!

closed 2021-04-26 by galluccib