prusa3d/PrusaSlicer #6439: MacOS export to webdav M1 silicon

pacaj2am, 2021-04-26:
### Version
Version 2.3.1+arm64

### Operating system type + version
PrusaSlicer
Version:   2.3.1+arm64
Build:     PrusaSlicer-2.3.1+arm64-202104161350

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Big Sur Version 11.2.3 (Build 20D91)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 Metal - 71.0.7
Vendor:       Apple
Renderer:     Apple M1
GLSL version: 1.20
Installed extensions:

GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
[... 343 lines cut ...]
com.apple.driver.AppleEmbeddedAudioLibs	1.17
com.apple.driver.AppleFirmwareUpdateKext	1
com.apple.driver.AppleM2ScalerCSCDriver	265.0.0
com.apple.iokit.IOSurface	289.3
com.apple.driver.IODARTFamily	1
com.apple.security.quarantine	4
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.kext.CoreTrust	1
com.apple.security.AppleImage4	3.0.0
com.apple.iokit.IOCryptoAcceleratorFamily	1.0.1
com.apple.driver.AppleARMPlatform	1.0.2
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IOReportFamily	47
com.apple.kec.pthread	1
com.apple.kec.corecrypto	11.1
com.apple.kec.Libm	1

** Stackshot Succeeded ** Bytes Traced 324430 (Uncompressed 820768) **
```

#### Project File (.3MF) where problem occurs
Not needed, occurs with any model

bubnikv, 2021-04-26:
Please contact your operating system vendor.

closed 2021-04-26 by bubnikv

pacaj2am, 2021-04-27:
In case someone is interested, luckily it has been resolved with 11.3 update