prusa3d/PrusaSlicer #6440: [BUG] Cutting fonction is broken on 2.3.1RC (maybe Mac M1 related?)

BastR, 2021-04-26:
### Version
v 2.3.1+arm64
Mac mini M1
 
### Behavior
The cut model function causes the stl to be destroyed, so far none of the stl's I have cut have given a good result.

https://user-images.githubusercontent.com/20129420/116073965-603cd400-a691-11eb-87b3-b27ab63d8174.mp4

rtyr, 2021-04-26:
Please provide some example with 3MF project file.

BastR, 2021-04-26:
[3MF.zip](https://github.com/prusa3d/PrusaSlicer/files/6379168/3MF.zip)

I provide 2 3MF project files, the first is with the original stl, the second with the cutted stl.

I tried with 20 other random stl, if the part is complex the STL becomes unusable.

If I insert a basic shape included in PrusaSlicer by right click, I can cut the parts without having any error

rtyr, 2021-04-27:
Does it happen if you repair the original model first? (it is broken). I was not able to reproduce it on win.

Repaired stl:
[Clip anti buée_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/6385948/Clip.anti.buee_fixed.zip)

rtyr, 2021-04-27:
I know, but original model is not OK (lot of errors), even though it is sliced correctly. Could you try the model I attached?

BastR, 2021-04-27:
I asked a friend to reproduce it on another mac M1 but with PS 2.3.0, the cutting was done without problems with the original STL (not the fixed version), so I think something is broken since version 2.3.0

[image]

I also did some other tests using the well known XYZ cube but also generating a basic shape directly in prusaslicer. The defects after the slice are different depending on the orientation of the cylinder

https://user-images.githubusercontent.com/20129420/116321239-fff58180-a7b9-11eb-818a-44300d10d9f7.mp4

BastR, 2021-04-28:
More test with an another friend with Mac Intel and the last version of PrusaSlicer. The cutting fonction works as expected, so it's really Mac M1 related @bubnikv 

[image]
[image]

trashcandrummer, 2021-05-09:
> Does it happen if you repair the original model first? (it is broken). I was not able to reproduce it on win.

To answer that question: No, repairing the STL does not solve the problem.

naghuid, 2021-05-30:
I'm had the same issue on my M1 Mac. PrusaSlicer is a universal binary so I tried the intel version (Get Info on PrusaSlicer.app and selected "Open using Rosetta") and that worked OK. So the issue seems specific to the native M1 build.

FidelCapo, 2021-09-16:
This issue should be fixed in PrusaSlicer 2.4.0-alpha1.
Closing.

closed 2021-09-16 by FidelCapo