prusa3d/PrusaSlicer #6442: Misc. glitches & feature requests.

amar-s, 2021-04-26:
Version 2.3.0+

_Use `About->About PrusaSlicer` for release versions_
^^^
**First Quirk: On Mac OS, the "About Box" should be the first item in the *Application* ("PrusaSlicer") menu. Having it under the "Help" menu is not according to Mac UI guidelines.

Similarly, "Check for Updates" should go under the application menu, NOT the "Configuration" menu.**

Mac OS 10.14.6

_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Original Prusa i3 Mk 3.

### Behavior

**One glitch: I have the skirt turned off (by setting skirt layers to 0). I noticed that the slicing report states 2 minutes print time for the non-existent skirt. This, of course, is incorrect. I believe that what is reported is actually for the skirt AND BRIM. If so, the labeling should be changed to reflect this.**

_Is this a new feature request?_

**Yes, partly.

1. When I specify a raft, PrusaSlicer makes it fill the entire outer outline of the part. In other slicers I have used, it only makes the raft to a certain perimeter around where the part contacts the build plate (similar to how the brim works). I would like the ability to do this in PrusaSlicer, but don't think there is currently a way to do this.

2. Also, when I specify support material be used, PrusaSlicer lays down greatly-expanded first layer(s) for the support material. I don't understand why they are so large. I would like a way to control this, so not so much material is wasted under the supports (and this material seems to interact with the brim in strange ways).

3. With other slicers and printers, I was used to using rafts, since they provided good bed adhesion, and simply peeled away to produce a very clean part bottom. I have had trouble getting such results with PrusaSlicer & i3. I may need to play with settings more to see if I can achieve this.

4. It seems an odd choice to have support material turned off by default. I think it should be turned on by default. Or at least you should get a warning if it is needed, but not turned on.

5. It seems that Prusa's preferred method of achieving better bed adhesion is to use a brim rather than a  raft. This works well for adhesion, but I find it difficult to easily and cleanly remove the brims from printed parts. I would like a way to increase the spacing between the part and the brim so that it may be more easily and cleanly peeled away from the part. I don't find a way to currently do this. (Separation can only be specified for support/raft material in the vertical direction.) I've played with the elephant foot setting to see if I could make things work better, but it doesn't help much and makes bed adhesion worse, defeating the whole purpose of the brim.**

rtyr, 2021-04-27:
> One glitch: I have the skirt turned off (by setting skirt layers to 0). I noticed that the slicing report states 2 minutes print time for the non-existent skirt. This, of course, is incorrect. I [...]

We will likely rename it.

> When I specify a raft, PrusaSlicer makes it fill the entire outer outline of the part. In other slicers I have used, it only makes the raft to a certain perimeter around where the part contacts the [...]

Raft improvements are already implemented on master. It will be included in PS2.4.0.

[image]

> Also, when I specify support material be used, PrusaSlicer lays down greatly-expanded first layer(s) for the support material. I don't understand why they are so large. I would like a way to control [...]

Also implemented on master. It will be possible to control this in PS2.4.0.

> With other slicers and printers, I was used to using rafts, since they provided good bed adhesion, and simply peeled away to produce a very clean part bottom. I have had trouble getting such results with PrusaSlicer & i3. I may need to play with settings more to see if I can achieve this.

It was reworked for PS2.4.0, bridge flow will be no longer used for the first layer on the raft, see https://github.com/prusa3d/PrusaSlicer/pull/6003#issuecomment-785010896.

> It seems an odd choice to have support material turned off by default. I think it should be turned on by default. Or at least you should get a warning if it is needed, but not turned on.

I don't agree. Some warning would be nice though (I believe there is already opened issue for that).

> I would like a way to increase the spacing between the part and the brim

That is what elephant foot compensation currently does. Brim offset will be added to PS2.4.0 though.

Please don't combine different topics in one issue, it is difficult to track.

rtyr, 2021-04-27:
> One glitch: I have the skirt turned off (by setting skirt layers to 0). I noticed that the slicing report states......

Label renamed to Skirt/Brim.

> **First Quirk: On Mac OS, the "About Box" should be the first item in the Application ("PrusaSlicer") menu. Having it under the "Help" menu is not according to Mac UI guidelines.

Same as https://github.com/prusa3d/PrusaSlicer/issues/3932.

> Similarly, "Check for Updates" should go under the application menu, NOT the "Configuration" menu.

It checks configuration updates, not application updates.

Closing.

closed 2021-04-27 by rtyr