prusa3d/PrusaSlicer #6443: Feature request: apply texture to outer perimeter of prints

DaveTSG, 2021-04-27:
Stefan of CNC Kitchen recently published a video in which he showed off a new feature in IdeaMaker slicer software - applying a texture to the outer walls of prints. The video is [here](https://www.youtube.com/watch?v=3-ygdNQThAs).

The examples shown in the video show how different textures can improve the stiffness of printed parts and prevent warping, even with large ABS prints printed without an enclosure. 

It seems to work by allowing the user to import an greyscale image of a pattern, and then horizontally offsetting the outer wall of the print based on how dark along the grey scale each pixel (within the image) is. While the textured walls can be designed in CAD manually, it would be a lot easier for the end user if this was implemented in the slicer.

Thanks for considering this request.

murk-sy, 2021-04-30:
I'm a bit torn if this should be:
 - a (separate?) tool that modifies the actual model (for easier reproducibility and transfer of data)
 - a tool within the slicer itself

And frankly, which it should be sort of depends on what you want to do with it.

For mechanical parts that feature partial decoration, reinforcement or even something like part number marking it should absolutely be a tool that modifies the actual model. That way you don't rely on the feature changing at some point to reproduce the same part, especially if fitment of 2 textured parts is a requirement.

For decorative parts, it would likely be most efficient in the slicer itself, especially if you do something like decorate easter eggs (which is something Angus of Maker's Muse played around with). That way you have an easy way of changing up what would otherwise be a boring collection of identical parts.

Ideally for both cases, Prusaslicer should just be able to export the modified model. That would allow it to pull double duty for stuff like adding a grip surface to a handle and things like that and then applying that so you have the effect reproducible.

Having this feature would also enable some now unexplored functionality - for example, being able to generate serial numbering, 2d barcodes and customer's logos and text on the prints? Readability experimentation for barcodes aside, *that* would be **fancy**.

KimmoHop, 2021-05-02:
I hope same file format for textures would be used (JSON with embedded PNG) - that would benefit both slicers.
[image]

Maybe the texture can be saved to 3MF file?

**PrusaSlicer special**: paint textured areas / not textured areas in the same way as supports and seams - the paint tool already exists :D

jrmoserbaltimore, 2021-05-22:
Would like to see this as well.  Also going to +1 for interoperability to avoid fragmented asset bases where we all go hunting around and just find things for the other tool.

RAMilewski, 2021-07-11:
I discovered I've been really spoiled by Prusa Slicer.   Switching slicers is painful.

 I figured I would just solve the problem by using Ideamaker for one print...  but then I hit the learning curve for ideamaker, and discovered there aren't any Prusa Mini profiles for Ideamaker to be found.

+1 for being able to export the model with the texture applied.  I couldn't figure out how to do that in Ideamaker.
+1 for @jrmoserbaltimore  comment too.

RAMilewski, 2021-07-11:
> Having this feature would also enable some now unexplored functionality - for example, being able to generate serial numbering, 2d barcodes and customer's logos and text on the prints? Readability experimentation for barcodes aside, _that_ would be **fancy**.

...and doing a color change would enable printing _very_ weather resistant barcodes and QR codes.

Qualith, 2021-09-27:
I watched a similar video from Maker's muse and I instantly imagined the potential for miniature scenery by just adding some texture.
I tried to make the same process of Ideamaker with blender and it's very much harder. 
Would be nice to have a simple Prusa's way tool to make it now that the slicer starts to tinker meshes.

Mike8-I, 2022-12-03:
I would also really love to have this feature. I have heard arguments that this should belong in the CAD software.
Though this might be right, to my knowledge it is not there in the CAD softwares that I know. It would be a great benefit for the surface finish to have it. Using textures can improve the surface finish and the look of objects significantly.

foreachthing, 2022-12-05:
> I tried to replicate the Ideamaker process with blender and it's very much harder. 

Not that hard, I think.
See file attached (remove .txt extension): [Cubes.blend.txt](https://github.com/prusa3d/PrusaSlicer/files/10158369/Cubes.blend.txt)

Granted, these are simple objects. It should just show what Blender can do (and it can do much, much more than this):
[image]

If exported via 3D-Printer addon, these files will be around 500 MB in size... just to note 😄

DaveTSG, 2022-12-21:
So Mikolas Zuza just showed off a new feature for PrusaSlicer v2.6 in the latest Prusa Live video (#48, at [16mins into the video](https://youtu.be/Rg8cHbahQYs?t=963)). This new feature allows adding text onto the surface of a printed object.

Using this feature, it may be possible to use a series of capital i letters ("I") to form vertical lines, and then use the "Z-rot" angle shown in the video to rotate it 90 degrees. The end result would be a series of horizontal lines jutting out from the object. There is also a depth setting (which seems to specify the thickness of the lettering) - it may be possible to specify a negative depth setting to cut into the object.

This may be a viable work-around until the feature is implemented properly.

petruut, 2024-03-25:
This would be such a useful feature, I have no idea why there’s no active development behind it…

murk-sy, 2026-09-24:
Still relevant as of 3.0.0-alpha12

Would ideally be implemented similarly to how fuzzy skin painting works

Current workaround: https://github.com/CNCKitchen/stlTexturizer