prusa3d/PrusaSlicer #6444: Mac M1 - Still freezing with 2.3.1

stefanpieter, 2021-04-27:
Version:
2.3.1

System:
Mac Mini M1 16 Gbit RAM

Issue:

The latest update from 2.3.0 to 2.3.1 has improved the freezing / crashing of PrusaSlicer but when selecting view flowrate visualiser (post slicing) the system crashes again just like in 2.3.0 with the CPU having a hard time. No matter the waiting time the program never recovers. It needs to be "force quit" to stop.

https://user-images.githubusercontent.com/65555837/116250405-9b103c00-a765-11eb-8876-904b7503a711.mov

bubnikv, 2021-04-28:
Would you please provide your 3MF and a precise sequence of steps to
reproduce your issue? We have an 8GB Macbook Air with the M1 CPU to test
with.

stefanpieter, 2021-04-28:
I was unable to save the original file in the video as it cashed but I have recreated the scene.

Steps taken:
1. Loaded STL files from a folder
2. Loaded more STL models from Fusion 360
3. Slice - no problems
4. Preview - no problem
5. Then, clicked View > Volumetric flow rate 
6. Wait....... a few hours later.... Force Quit
7. Cry in my hands

[Issue#6444.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/6394493/Issue.6444.3mf.zip)

DumbGameMaker, 2021-05-01:
having the same issue loading a large g-code file.

mattapperson, 2021-05-26:
Same issue here.

closed 2026-09-29 by prusaslicer-bot[bot]