prusa3d/PrusaSlicer #6445: generated lines much thinner than expected

cbc02009, 2021-04-27:
### Version
Version 2.3.1+win64

### Operating system type + version
Windows 10 20H2 19042.948

### 3D printer brand / version + firmware version (if known)
Voron 2.4 with Klipper firmware

### Behavior
I'm trying to manually create a profile for my Voron 2.4 based off the Super Slicer profile I've been using for a long time, but the line widths in Prusa Slicer are not being generated as expected.

Take this picture:
[image]

both are using .2 layer height with a .4 line width and identical extrusion multipliers. The one on the left is Prusa Slicer, and the one on the right is Super Slicer. Both show the same layer height, both models are exactly the same size in PS and SS, but the line widths are completely different. The print times are also wildly different: 2h30m for SS and over 6h for PS.

I've attached the project file, my settings bundle, and the g code generated by Super Slicer for comparisons sake.. Any help getting this set up would be greatly appreciated, as I would love to try out Prusa Slicer.
[[a]_Foot_ALL.zip](https://github.com/prusa3d/PrusaSlicer/files/6386302/a._Foot_ALL.zip)
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/6386352/PrusaSlicer_config_bundle.zip)
[[a]_Foot_ALL_0.2mm_ABS_2h38m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/6386381/a._Foot_ALL_0.2mm_ABS_2h38m.gcode.zip)

neophyl, 2021-04-27:
I knew what it would be without looking at your project.  I did however open it to confirm :)  While the programs share most things in common it is a mistake to think they are still fully compatible.

With extrusion widths set to % values PS computes the widths based on LAYER HEIGHT.  Says so in the tool tip.  
Super Slicer however due to popular demand changed it to be based on Nozzle diameter.  Also says so in the tool tip.

The request has been made to make it work that way in PS, but of course this would be a breaking change for some users profiles.  Which plays into the decisions made by Prusa.  
Super Slicers user base is much smaller, it also tends to be those who have a bit more experience to venture off the beaten track and like taking a few more risks lol.  As such it wasn't so much of a risk to change it.

cbc02009, 2021-04-27:
> I knew what it would be without looking at your project. I did however open it to confirm :) While the programs share most things in common it is a mistake to think they are still fully compatible. [...]

Yup. Read too fast. Thanks!

closed 2021-04-27 by cbc02009