prusa3d/PrusaSlicer #6446: created a new printer profile for 0.8mm nozzle, but now cannot change filament settings

Gdipstick, 2021-04-28:
PrusaSlicer   ver. 2.3.1 +win64

windows  / Chrome Version 89.0.4389.128 (Official Build) (64-bit)

MKS3+

Created a new pritner profile for a new 0.8mm nozzle.   But now when i try to change or add a new filament setting the new setting does not save

rtyr, 2021-04-28:
Could you be more specific about your workflow please?

> Created a new pritner profile for a new 0.8mm nozzle

Created manually or added through configuration wizard? (profiles for 0.8mm nozzle are bundled in PrusaSlicer)

> But now when i try to change or add a new filament setting the new setting does not save

How are you adding a new filament profile? What is the exact name of the filament profile you want to add?

WoLpH, 2021-04-30:
I can confirm this. Not sure in which situations it does and does not happen, but only PVA+ is showing up for my 0.8mm printer profile

> Created manually or added through configuration wizard? (profiles for 0.8mm nozzle are bundled in PrusaSlicer)

Created through the configuration wizard.

> How are you adding a new filament profile? What is the exact name of the filament profile you want to add?

Adding it through the wizard. It's just not showing up.

Running the wizard:
[image]

Result:
[image]

I believe it's due to PrusaSlicer sometimes (not sure how/why) deselecting the Printer. It has disappeared again for me:
[image]

After re-adding the 0.8mm printer it works again

rtyr, 2021-04-30:
@WoLpH Filament profiles you have selected have it's specific variant for 0.8mm nozzle (for example `Prusament PETG @0.8 nozzle`). 0.8mm specific filament profiles are installed only if 0.8mm nozzle variant is enabled. I am not sure what can cause the behavior related to your last screen (it is stored in PrusaSlicer.ini file).

rtyr, 2021-04-30:
I guess that @Gdipstick maybe created his custom 0.8mm printer profile manually, so 0.8mm filament profiles were not enabled/installed by the configuration wizard.

WoLpH, 2021-04-30:
> @WoLpH Filament profiles you have selected have it's specific variant for 0.8mm nozzle (for example `Prusament PETG @0.8 nozzle`). 0.8mm specific filament profiles are installed only if 0.8mm nozzle variant is enabled.

That makes sense

> I am not sure what can cause the behavior related to your last screen (it is stored in PrusaSlicer.ini file).

I am guessing that it's due to multiple instances of  PrusaSlicer. When you close an instance it writes its settings and therefore overwriting (and removing) the printer selection. When you open the 3mf file it will show the printer but it won't be enabled so you don't see the filament.

rtyr, 2021-09-08:
No response. Closing.

closed 2021-09-08 by rtyr