prusa3d/PrusaSlicer #6448: Seam management/tuning

Lindsay-Wood, 2021-04-28:
### Version
2.3.xxx

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win10 64
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S/current PS firmware
Ender 3/marlin
Ender 3/klipper
Voron 2.4/klipper

### Behavior
I'm consistently getting under-extruded or "puckered" seams from PS across a range of printers and firmwares. All printers have pressure advance tuned correctly, and I've performed pressure advance (or linear advance for marlin) tuning towers on the parts in question with no improvement in seam quality. Currently mainly printing in PLA but seems endemic to rigid filaments.

See photos

What I've tried (generally with a tuning tower sweep in klipper or similar):

- Retract length tuning from 0-2mm, doesn't produce significant improvements (DD)
- Lift Z/Z-hop/no lift Z
- Pressure advance tuning/sweep- running pressure advance tower on the part in question, no major improvement in seam
- Extruder tension adjustment
- Seam positioning (of course sharp corners help hide it, but that's not always an option

My working theory is with well-tuned printers (tuned with input shaper/pressure advance on the klipper printers, and the very reasonable factory prusa tuning) that the printer is following the slicer instructions too well-i.e. starting/stopping with a small gap between the path endpoints, leaving effectively two circles butted against one another. This produces the puckered seam (as opposed to a printer without PA or proper tuning, which might ooze a blob). I would have expected "extra length on restart" to help with this, but I haven't seen good results. I'm hoping that a "perimeter end overlap" option might be possible, eg. you butt the ends up an additional 10% or so to help close the seam. I haven't found such an option, even in SS and similar (superslicer allows you to change perimeter overlap, but this is between perimeters, not at the start/end.

Any ideas? I'm surprised there aren't more techniques (that I've found) to manage this
[image]
[image]
[image]
[image]

_Is this a new feature request?_
Maybe? Unless there's existing management strategies 

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Lindsay-Wood, 2021-04-28:
Okay, I have a theory:

[image]
This is the default PS behaviour, at the end of a outer wall extrude it travels/wipes inwards, which could cause the divot as shown in a well setup printer

My assumption was that adding wipes would prevent this, but that doesn't seem to be the case, as the slicer kicks inwards, then back out to do the wipe- potentially exacerbating this issue:
[image]

Well, let's compare other slicers in a similar scenario:

Cura has the option for "outer wall wipe distance" which makes the extruder travel along the path, without any extrusion- it does not kick inward like PS:
[image]
(ignore the blue line going inwards, that's from the start of the extrusion move)

Now, superslicer. I wasn't expecting a useful result out of superslicer as it's a very similar PS fork, and initially it looked like I was right:
[image]
It kicks inwards exactly like PS does- not surprising

However I noticed SS has an additional option "extra wipe for external perimeters", similar to Cura:
[image]

This causes the extruder to travel tangent to the path without performing the PS style dip inwards first, after the wipe motion it then wipes inwards:
[image]

Any thoughts? I'll test this out this afternoon. Hopefully I don't sound too crazy

rtyr, 2021-04-28:
"extra wipe for external perimeters" 0/2mm/3mm. It doesn't look like an improvement.
[image]

Lindsay-Wood, 2021-04-28:
> "extra wipe for external perimeters" 0/2mm/3mm. It doesn't look like an improvement.
> [image]

Cura or SuperSlicer?

rtyr, 2021-04-28:
SuperSlicer

Lindsay-Wood, 2021-04-28:
Interesting. Shame, I thought that might be the smoking gun.

rtyr, 2021-04-28:
It was printed with disabled LA though, if I remember correctly (there was some bug in SS at the time when I tested it). Plus I think this can cause insufficient pressure in the nozzle.

Lindsay-Wood, 2021-04-28:
I'll give both Cura and SS a go this afternoon on unified settings and then experiment with this and see what the results are like.

I went away from SS to PS as although SS had more features, I also found bugs with it more often (for example the current version crashes if I select the same printer profile again, in order to clear edited settings). PS is maintained and caters to a larger audience, so it seems to be a bit more robust, and most bugs are solved faster.

ericderace, 2024-01-03:
@Lindsay-Wood Did you manage to improve your seams in the end? I just started trying to tune for the same ting (OrcaSlicer in my case) - Tried a bunch of things - no improvements.

closed 2026-09-29 by prusaslicer-bot[bot]