prusa3d/PrusaSlicer #6450: Is eLut.exe an Trojans ?

MasseWeg, 2021-04-28:
Hello all,

I download the PrusaSlicer 2.3.0 Version as source code to compile it in MS VisualStudio. 

At first, thanks a lot to the people who made the manual for the source installation, it works very good :) 

So now to my problem.

My Kaspersky recognized a Trojan in the Installation-Files. 

Its on path: 
C:\PrusaSlic3r\src\PrusaSlicer-version_2.3.0\deps\build\dep_openexr-prefix\src\dep_openexr-build\IlmBase\Half\Release

See the Image what I have attached.

[image]

Is eLut.exe an Trojan or not. What does this .exe do ?

The PrusaSlicer works on VS without any problems although my Kaspersky put the eLut.exe in quarantine.

Many thanks in advance for the upcoming help.

greetings 

MasseWeg

bubnikv, 2021-04-28:
I suppose eLut.exe is some helper application compiled by openexr library,
which may or may not be required to compile PrusaSlicer. I doubt it is a
trojan.

MasseWeg, 2021-04-28:
Hello bubnikv,
thanks for the fast answer. 
I also doubt that it is an Trojan. If it is, it would be a bit strange that I m the first who ask for :) 
I expressed myself wrongly in the previous post.
I allowed my Kaspersky to ignore it, that the source "installation" will work fine and all went good. 
Some days later I had a blue screen. After 3 of them with most 2 days between them, I allowed Kaspersky  to move the eLut.exe to quarantine again. I haven't had a blue screen since then and can use PrusaSlicer without any problem.

So that's why I'm a little unsure at the moment.

rtyr, 2021-05-03:
Can we close this issue?

closed 2021-05-26 by MasseWeg