prusa3d/PrusaSlicer #6455: PrusaSlicer 2.3.1 installer doesn't run on Ubuntu Linux

rgezikov, 2021-04-29:
### Version
PrusaSlicer 2.3.1.

### Operating system type + version
roman@roman-laptop:~/tmp/prusa3d_linux_2_3_1$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
On staring PrusaSlicer-2.3.1+linux-x64-202104161339.AppImage a splash screen appears for an instance and disappears.

Output to the terminal:
roman@roman-laptop:~/tmp/prusa3d_linux_2_3_1$ ./PrusaSlicer-2.3.1+linux-x64-202104161339.AppImage 
The program 'prusa-slicer' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 611 error_code 2 request_code 154 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

rgezikov, 2021-04-29:
More info: the binary that previously ran fine doesn't start anymore either with the same output. It happened after latest update with sudo apt upgrade.

AndKe, 2021-06-03:
Already existting, previously working PrusaSlicer won't start anymore on Ubuntu 21.04 - I suspect that happened after upgrading from 20.10

lukasmatena, 2022-04-05:
Is this issue still valid with latest system packages and PrusaSlicer 2.4.1?

AndKe, 2022-04-05:
no, 2.4.1 is good now.

lukasmatena, 2023-02-06:
Thanks, closing then.

closed 2023-02-06 by lukasmatena