prusa3d/PrusaSlicer #6461: Feature Request: Z seam in specified XY coordinates and/or "sharpest corner" Z seam option

kr15uk, 2021-04-29:
### Version
2.3.1 ARM64

### Operating system type + version
macOS BigSur ARM

### 3D printer brand / version + firmware version (if known)
Ultimaker

### Behavior
Ability to have forced straight vertical Z seam in specified XY coordinates.
Or like in Cura "sharpest corner" Z seam would be a dream.
As a workaround I do rear Z seam and rotate all objects 45degree to have the seam in one corner, although often it's neither straight nor all there in full.

I believe aligned is suppose to be "sharpest corner" equivalent but nope, not even close.

Paint on Z seam feels like a gimmick sorry, maybe for organic shapes it does work but not for mechanical fundamentally square objects.

#### Project File (.3MF) where problem occurs

rtyr, 2021-04-29:
I think it would be better to implement something like this https://github.com/prusa3d/PrusaSlicer/issues/5970, https://github.com/prusa3d/PrusaSlicer/issues/6242, instead of XY coordinates.

murk-sy, 2021-04-30:
Since for mechanical parts you're usually making boxes or brackets, generally things that have at least 1 hidden surface that's attached to something, having seams bunched towards a particular corner allows much faster print preparation for tests. Having a position dropdown with all ordinal directions just listed would already be a plenty (assuming it then picks the corner instead of position exactly at an angle from center).

Seam painting is great (in concept, I'm having some issues with it, see #6400), but having to seam paint every single model would be a chore.

I've had aligned give me fairly annoying intermittent gaps on the front of the model, and I'm having some seam painting issues, so my only way to reliably print nice looking parts is to rotate them and use Back alignment, which reduces the usable print bed area, sometimes dramatically, and changes the top/bottom infill direction.

towlerg, 2021-05-05:
"Paint on Z seam feels like a gimmick sorry, maybe for organic shapes it does work but not for mechanical fundamentally square objects."

Perhaps if the the cursor could be constrained horizontally and vertically it would feel like a "gimmick"

murk-sy, 2022-08-10:
With 2.5.0 alphas and improved seam positioning, this seems mostly resolved, though I still get some pretty weird results sometimes - but so far I'm pretty happy with it.

Really01101001, 2023-04-09:
Yes X,Y user defined coordinates like Cura and ideaMaker have. "Painting" a Z seam? I'm trying to do work beyond 3rd grade art class...

meteozond, 2026-09-23:
This is really needed :(