prusa3d/PrusaSlicer #6462: Buggy support generation on holes

batata004, 2021-04-30:
Hi, PS is awesome, the best slicer I've ever tried but support generation is one of the worst I ever tried. This time, my problem is using support on big holes printed vertically. Take a look at this part:

[image]

When I slice it with supports enabled, look what happens:

[image]

This is completely unnecessary, waste of material, waste of energy, of time... Other slicers would only create supports inside the holes themselves, why is PS creating a support from the build plate? I should only create a support from the bottom of the hole till the top, supporting the top part of the hole. Hope this helps improve this software!

rtyr, 2021-04-30:
Fixed with af9c7c9. Closing.

```
Implementing a new switch for the shape of support towers:
expanded to a grid (the old way) vs.
snug (like the upstream Slic3r, Cura or Ideamaker).

Snug supports suffered from the degeneracies when merging overhang islands
over a large number of layers when projecting the support towers down.
We borrowed the idea & a bit of code from Cura by simplifying the support
polygons by closing the concave cracks, see the smooth_outward() function
and the MutablePolygon class.
```

closed 2021-04-30 by rtyr

batata004, 2021-04-30:
@rtyr I installed the latest version of prusa and I cant see the snug option, please see print below:

[image]

rtyr, 2021-04-30:
It is implemented on master. It will be available in PS2.4.0.

[image]

batata004, 2021-04-30:
Thank you so much!!