prusa3d/PrusaSlicer #6465: More problems with supports.

Despx72, 2021-04-30:
### Version
2.3

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Ender 5

### Behavior
Hello,
I print very small objects (layer 0.06 mm nozle 0.2 mm petg).
I have many problems with supports's management because PrusaSlicer only manages two Z offsets between the support and the layer bridge 0.2 mm and 0 mm (soluble).
I also find that the 0.2 mm setting never respects this value but usually the distance is 0.4-0.6 mm and for my objects too far away.
If I write values smaller than 0.2 mm (for example 0.1 mm) in the Suppot_Material_Contact_Distance box, the distance is not reduced.
I ask you how I can have a real distance between the support and the bridge layer 0.1 mm yes space.
I would also like to ask you how I can enable Support_Material_Synchronize_Layers even when the value in Suppot_Material_Contact_Distance is greater than 0 because I have to be sure that the height of the support is identical to that of the object to better manage the Z offset between support and object.

Thanks so much for your reply.
Despx

rtyr, 2021-04-30:
> I have many problems with supports's management because PrusaSlicer only manages two Z offsets between the support and the layer bridge 0.2 mm and 0 mm (soluble).

You can use your own custom value, not only predefined 0.2/0. It is also possible to use negative values for experimenting. 

> I also find that the 0.2 mm setting never respects this value but usually the distance is 0.4-0.6 mm and for my objects too far away.

Bottom most layer of the supported object is printed using thick bridge flow (round extrusion of the same diameter as nozzle diameter). So 0.2mm nozzle diameter + 0.2mm z contact distance = 0.4mm. See #102 and other linked issues for more information.

We have recently added a new option ("thick bridges"), that will allow to optionally disable standard thick bridges. This option will be available in PS2.4.0.

`To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio.`

> I would also like to ask you how I can enable Support_Material_Synchronize_Layers even when the value in Suppot_Material_Contact_Distance is greater than 0 because I have to be sure that the height [...]

PrusaSlicer prints the support layers independently from the object layers by default, meaning that the object layers are printed at layer heights independent from the support layer heights. This is an effective strategy if the object / support gap is not a multiple of the object layer height, and this strategy also allows the support columns to be printed with thicker layers thus faster than the object layers. Synchronized layers can be currently enabled only for soluble supports (contact Z distance = 0).

Despx72, 2021-04-30:
I thank you for the answer.
The problem in my case is that I print very small objects and if the layers of the supports do not follow those of the object it is difficult (or luck) to set the correct Z offset due to the discrepancy between the height of the layers between support and object in fact printing with 0.1 mm layer:

With 0.2 mm as the Z offset, I can have between 0.2 and 0.6 mm of distance.
With 0.1 mm as the Z offset, I can have between 0.2 and 0.4 mm of distance.
With -0.01 mm as the Z offset, I can have between 0.1 and 0.3 mm of distance.

Imagine what happens when I print with 0.06mm layers ... a disaster!
In short, if I set a Z Offset value it is not a fixed rule but a randomness that I get it in the slicer.
I imagine this condition may not give problems with large printed objects but when in an object ... 0.25 mm "makes the difference", the management of the supports must be very precise.
Could you in ver. 2.4.0 change that the Support_Material_Synchronize_Layers can be enabled / disabled optionally and not only by setting value = 0?
(If I don't have a dual extruder printer ... I don't do anything with it ;-)
Otherwise, it would be a good thing if Z offset of supports were handled as in SuperSlicer which is much more precise than PrusaSlicer.

Thank you very much.
Despx

Despx72, 2021-05-11:
I noticed that PrusaSlicer incorrectly calculates the Z offset between the supports and the object.
If with layer 0.06 mm I set the value of 0.2 mm of z offset I get a distance between the last layer of the support and the bridge layer of the object of almost 0.6 mm which for my prints is absurd. A correct value should be about 0.2 - 0.36 mm.
I understand that most people print very large objects and don't care about half a millimeter but there should be better accuracy from PrusaSlicer.
Another thing, why the synchronize-with-object-layer function can only be used with Z offset = 0 mm ??
It would be very convenient to have greater precision in the calculation of the Z offset. Is it not possible to unlock this function also for Z offset greater than 0 and leave the user the right to use it or not?

Below you can see the calculation error of the z Offset of the supports:
value set in the program 0.2 mm
value obtained by the slicer 0.59 mm!!

[image]
[image]

Here a sample project:

[Test_Wrong-Supports_001.zip](https://github.com/prusa3d/PrusaSlicer/files/6456989/Test_Wrong-Supports_001.zip)

I have seen in PRUSAPRINTERS FORUM that I am not the only one having problems with supports management and these two topics complain about the same problems as mine:

https://forum.prusaprinters.org/forum/prusaslicer/prusa-slice-2-3-supports-problem/
https://forum.prusaprinters.org/forum/prusaslicer/bad-support-surface-bug-or-wanted-w-picture-josef-look/

It is really frustrating to see all the hours spent designing a 3D model in the best possible way ruined during printing due to a stupid bug.

Best Regards 
Despx

rtyr, 2021-05-11:
It is not a bug, it is just how Slic3r/PrusaSlicer works and always worked. Please read my previous answer. 

> Below you can see the calculation error of the z Offset of the supports:
> value set in the program 0.2 mm
> value obtained by the slicer 0.59 mm!!

That is correct (with current implementation). The bridge (blue part) is not printed with the flow equal to the 0.06mm layer height, it is printed with the bridge flow (0.4mm extrusion in this case). Imagine the 0.4mm thick extrusion layed down at Z2.58. You will get roughly 0.2mm Z-contact distance from the supports. This approach is specific to Slic3r/PrusaSlicer (and SuperSlicer, it just gives you a way to specify whether set distance is from the filament or from the plane).
`0.2mm Z-contact distance + 0.4mm bridge (equal to nozzle diameter) = 0.6mm`.

I can understand that the way PrusaSlicer handles this is confusing and hard to understand. It is also questionable, whether this approach inherited from upstream Slic3r is ideal.

As I mentioned earlier, in the upcoming PS2.4.0, there will be a new option, which will allow to disable those thick bridges. With the thick bridges disabled, it will behave like you expect. We will likely disable thick bridges by default in our print profiles.

rtyr, 2021-05-11:
I would only repeat myself. See the images below, it might help you to understand how it works and why it is directly related to the thick bridge layers... It is sliced with 0.2mm layer height and 0.2mm z-contact distance. Notice the thickness of the blue line in relation to the normal layer height. Also notice the height, where the thick bridge line has to be printed to achieve the proper z-contact distance.

1) Thick bridges (current standard).
[image]

2) Regular flow (like Cura, Simplify, etc.)
[image]

> Make the Support_Material_Synchronize_Layers function activable by the user with Z offset greater than zero.

https://github.com/prusa3d/PrusaSlicer/issues/553. 

> Let the user decide the right distance between support and object without PrusaSlicer's automatisms.

It is what PrusaSlicer does, unlike most of the other slicers (it has to be a multiple of layer height in other slicers, like Cura).

Anyway. In the PrusaSlicer 2.4.0 you will have the option to choose between 1) (current behavior) and 2) (similar to Cura, Simplify, ...).

Closing.

closed 2021-05-11 by rtyr