prusa3d/PrusaSlicer #6466: Prusa Slicer fails to slice quite simple STL

bassklampfe, 2021-04-30:
### Version
Version 2.3.0+linux-x64
( but could also reproduce with PrusaSlicer-2.3.1+linux-x64-202104161339.AppImage )

### Operating system type + version
Ubuntu 18.04.5 LTS on i7-8086K CPU

### 3D printer brand / version + firmware version (if known)
n.a.
#
## Behavior
* attached Clamp-2.0mm.stl fails to slice. Clamp-2.1mm.stl - which is only minimal different slices well
* just load the STL and slice with 0.20mm QUALITY settings

Screenshots (with both STL's loaded)
[image]
[image]

Attached both STLs and the source OpenSCAD script to create STL's
Difference is if clampg=2.0 (fails) or clampg=2.1 (slices)
[PrusaSlicerDefect.zip](https://github.com/prusa3d/PrusaSlicer/files/6405847/PrusaSlicerDefect.zip)

lukasmatena, 2021-04-30:
I reproduced the problem in 2.3.0 but not in current master. I guess this was fixed by 9c5dd35, along with many other similar issues related to OpenSCAD models.

I'm closing the issue, if it persists in 2.4.0 (after it is released), let us know. Thanks.

closed 2021-04-30 by lukasmatena

bassklampfe, 2021-05-01:
Thank you for the update. I'll use clampg=2.001 as workaround and wait for the next release.