prusa3d/PrusaSlicer #6470: Model won't slice in Prusa 2.3.1

jrmoserbaltimore, 2021-05-01:
### Version
2.3.1 x86-64

### Operating system type + version
Windows 10

### Behavior
This model, scaled to 50%, will slice in 4 seconds in Cura; in PrusaSlicer, it stops at "Preparing infill…" for all eternity.

[layens-400-end-panel.zip](https://github.com/prusa3d/PrusaSlicer/files/6409756/layens-400-end-panel.zip)

rtyr, 2021-05-01:
Please provide your 3mf project file.

jrmoserbaltimore, 2021-05-01:
Here you go
[layens-400-end-panel.zip](https://github.com/prusa3d/PrusaSlicer/files/6409814/layens-400-end-panel.zip)

rtyr, 2021-05-03:
It is caused by Archimedean chords infill patern. The slicing will likely finish after some time.

jrmoserbaltimore, 2021-05-03:
That's interesting.  Also I'm using gyroid, I assume you mean the top fill?

ArcWelder turns those into a handful of arcs.  I wonder if there's a fast way to collision detect and split; but then, it's probably losing its mind on the text engraving, which is difficult and complex.

rtyr, 2021-05-03:
Yes, I meant top/bottom pattern.

bubnikv, 2021-05-04:
I hoped the issue was similar to the Hilbert curve issues we fixed in the
upcoming PrusaSlicer 2.4, but this issue is different. Namely, a dense long
spiral is being clipped with a simple curve and the Clipper library does
not like such configuration. We may end up splitting the long spiral into
multiple pieces, clipping it and then stitch it again.

closed 2026-09-29 by prusaslicer-bot[bot]