prusa3d/PrusaSlicer #6471: Minimizing PrusaSlicer on MacOS Big Sur

MSeeger-maker, 2021-05-01:
### Version
2.3.1+arm64-202104161350

### Operating system type + version
MacOS Big Sur 11.3 Build 20E232

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + MMU2S

### Behavior
When I minimize PrusaSlicer to my Dock, clicking on the icon in the Dock "activates" the application but doesn't make it reappear on the screen. In order to maximize the application again, I need to either force it closed and reopen it or right click on the icon, choose "Show All Windows," and find it that way. I'm not sure what other information is needed about the behavior, but it happens every time I minimize the application so it should be pretty easy to replicate.

I don't know if this was present in previous versions. This is the first version of PrusaSlicer I've installed on MacOS (thanks for making a native M1 version!).

#### Project File (.3MF) where problem occurs

radaiko, 2021-05-02:
+1 same for me after updating to 11.3. with 11.2.3 minimizing worked

rtyr, 2021-05-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6395. Closing.

closed 2021-05-03 by rtyr