prusa3d/PrusaSlicer #6472: Slicing Preview - Top Layer Slider Should be Selected by Default

werickson24, 2021-05-01:
### Version
_2.3.0_

### Operating system type + version
_Win 10_

### 3D printer brand / version + firmware version (if known)
_MK2.5s MMU2.0_

### Behavior
In preview, the bottom layer slider is auto selected. It should be changed so the top slider is selected by default. This was randomly changed and is annoying since the sliders are hard to click and its useless to be looking at the bottom of a model. 

_Is this a new feature request?_
Yup

rtyr, 2021-05-03:
Top layer is selected by default.

Can you provide exact steps to reproduce the issue?

werickson24, 2021-05-03:
Add any stl file, it shouldn't matter. 
Click "Preview" or "Slice Now", both do the same thing. 
Then hover over the layer selector and scroll, the bottom slider will be auto selected and move instead of the top one.

rtyr, 2021-05-03:
I am not able to reproduce it.

werickson24, 2021-05-03:
And it literally just fixed itself.

werickson24, 2021-05-03:
Thank you for the luck I suppose.

closed 2021-05-03 by werickson24