prusa3d/PrusaSlicer #6473: Double clicking on 3mf file doesn't set filename in PrusaSlicer

Sembazuru, 2021-05-02:
### Version
Version 2.3.1+win64

### Operating system type + version
Edition	Windows 10 Pro
Version	20H2
Installed on	‎2021-‎03-‎30
OS build	19042.928
Experience	Windows Feature Experience Pack 120.2212.551.0

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S (Not relevant to issue)

### Behavior
* When double clicking on a .3mf file in my filesystem the "input_filename_base" variable isn't set. See missing filename in the window's title bar.
[image]

* When opening the same .3mf file from within PruasSlicer the "input_filename_base" variable is set. See the filename in the window's title bar.
[image]

* Both methods of opening the same file should result in the "input_filename_base" variable being set.

#### Project File (.3MF) where problem occurs
This happens with all my .3mf files, but here is the file from the screenshot examples:
[cap_CenterCutout.zip](https://github.com/prusa3d/PrusaSlicer/files/6410623/cap_CenterCutout.zip)
FWIW, the original stl is from the PrusaPrinters repository https://www.prusaprinters.org/prints/32024-gyroid-pen-holder-with-knurled-screw-cap with my own PrusaSlicer customizations.

combolek, 2021-05-04:
This may be intentional? When you double click on a .3mf file this is the same as running prusa-slicer from the command line and specifying the .3mf file as an argument. This behaves like import model, not open project. For example you can specify multiple files on the command line and they will all be imported. Import object from the menu does not normally set the project name either.

Sembazuru, 2021-05-05:
Not quite the workflow I expect when double clicking on a file from my filesystem. I wonder if a modification of my initial request might work for both workflows. Maybe this?:

- If a single file is passed to PrusaSlicer when opening then open the file along with setting the project name.
- If multiple files are passed to PrusaSlicer when opening it imports the files w/o setting the project name.

As it is with the current implementation if I open a 3mf file (that I've built with all my custom settings, modifiers, etc) and immediately send it to OctoPrint, if I don't watch out then the g-code received by OctoPrint isn't named how I expect. This is quite a hassle for my workflow. But I can see (now that I understand more of the issue) how fundamentally changing how PrusaSlicer reacts to loading a file from the command line might break some existing scripts/batch files that expect the project name to not be set.

bubnikv, 2021-06-03:
Will be fixed with PrusaSlicer 2.4.0-alpha1.

closed 2021-06-03 by bubnikv