prusa3d/PrusaSlicer #6474: Arrange feature - instance print order/positioning improvement (re: Complete individual objects, enabled)

Jaxx005, 2021-05-02:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10 Pro N
Version 10.0.19042 Build 19042

### 3D printer brand / version + firmware version (if known)
Standard factory Prusa MK3S (purchased 2020).
Firmware version: 3.9.3-3556

### Behavior
1) Print setting >> Complete individual objects ("CIO") >> ENABLE   
2) Add instances (eg six of single body)
3) Arrange

The Arrange function correctly heeds the 'CIO' setting and spaces instances correctly...
However, the order/positioning on the bed seems 'unusual', not as expected (the logic is unknown/unclear)

The instances on the bed (eg six instances) were:  two rows (across bed) x 3 deep

The print order was:  
- front to back
- to left
- to front again...  
- etc.

### Problem
This is OK for short models that are below the wiring loom (at the rear of the printhead).
However, for *taller* models, this may cause the wiring to catch an already printed instance.

### Current work-around
The instances are numbered and printed in that order, so each instance needs to be re-positioned on the bed, to match the print-order above.

For example:

last row........ [6]  [5]   << 5th and 6th instances
next row...... [4]  [3] 
front row..... [2]  [1]  << first instance on the right

This manual process is awkward rather than complex, but:
1.  it 'voids' the Arrange function (which would obviously revert to the old print order!)
2.  requires the user to Preview in order see the print-order of instances
3.  requires the user to manual reposition instances, clear the collision warnings, and repeat until all OK.

### Suggested Improvement (&/or clarification of logic? )
Arrange (certainly when CIO enabled) should perhaps: 
1. print the front-most instances first (leaving the bed always clear behind)
2. then print from the righthand side (leaving the bed always clear to the left, reducing collision with sensor wires)

### Comments:
CIO and Arrange are brilliant features, but I think this change/option could make it more effective, safer and less wasteful too?

CIO has saved me many hours - especially when the last (e.g. 6th) instance is the one that fails near to its completion!

#### Project File (.3MF) where problem occurs
No 3MF file included - it's easy to replicate with any single object and multiple instances etc

### screen shots
Added 3 screenshots - showing manually positioned instances
[image]
[image]
[image]

Hope that all makes some sense! :)

Thanks.

Jaxx005, 2021-05-02:
### Additional/update:
Further screenshots - showing the order given by Arrange funtion:
Especially note that instance 5 and 6 (printed last) are in front of the other already printed instances, so rear wiring is likely to strike the instances 1 and 3 (second row back).   ££££ :(
[image]
[image]
[image]
[image]
[image]
[image]

echo465, 2021-12-31:
This is similar to https://github.com/prusa3d/PrusaSlicer/issues/486 - but with instances you can't drag/drop to change the printing order.

Jaxx005, 2022-01-02:
Yes, #486 (closed?) seems to be similar problem but for _Objects_, not _Instances_. 

The **Arrange** issue seems the same is both cases: both Instances and Objects are not positioned optimally/logically as outlined above?

lukasmatena, 2025-02-18:
Arrange and crash detection for sequential printing was improved in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). I believe it solves this issue. Your feedback is welcome. Closing.

closed 2025-02-18 by lukasmatena