prusa3d/PrusaSlicer #6478: Incorrect slicing - layers lost

abukharov, 2021-05-02:
### Version
Version 2.3.0+win64

### Operating system type + version
Edition	Windows 10 Pro
Version	21H1
Installed on	‎16/‎06/‎2020
OS build	19043.964
Experience	Windows Feature Experience Pack 120.2212.3740.0

### 3D printer brand / version + firmware version (if known)
Custom CoreXY
Board: Duet 2 WiFi (2WiFi)
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.2.2 (2021-02-11)
Duet WiFi Server Version: 1.25

### Behavior
I was slicing this model of a fidget toy from Thingiverse (STL is in the zip together with the 3MF) and looked at the bi-conical joints. Chunks of the inner part were missing in two of three joints. The third one was sliced correctly:
[image]
[image]

I exported the gcode and ran it through https://gcode.ws/ and it showed that the problem was with the actual gcode, not with rendering or something else:
[image]

The other slicer I have, generates correct gcode (and it also prints fine):
[image]
[image]

Here is the structure of the joints for reference:
[image]

#### Project File (.3MF) where problem occurs
Also the STL
[fidget.zip](https://github.com/prusa3d/PrusaSlicer/files/6412448/fidget.zip)

rtyr, 2021-05-03:
I am afraid I don't see any issue on your screenshots, maybe put some arrows where do you think is some issue. The model seems to have some errors, but it slices fine for me.

I can see there is some gap (so it can move), so I don't understand what is the issue on your screenshots. It would be also better to disable "detect bridging perimeters" in this case.

[image]

rtyr, 2021-05-03:
Oh I see it now, altough I am not able to reproduce it.

Use "Fix through the Nefabb" option, it will likely solve your issue.

[image]

abukharov, 2021-05-03:
Oh, I'm sorry. Fixing through Netfabb helps indeed. I normally rely on the orange exclamation sign to know that the model is broken, this time it was displayed as healthy and manifold, but in fact was broken.

Thanks very much for your time, I'm closing the issue.

closed 2021-05-03 by abukharov