prusa3d/PrusaSlicer #6481: Build Issue - Boost fails to download.

TheAndr0id, 2021-05-03:
### Version
2.3.1

### Operating system type + version
Linux x86

### Behavior
When trying to build using the build deps process, the boost 1.70 source fails to download:

```
wget https://dl.bintray.com/boostorg/release/1.70.0/source/boost_1_70_0.tar.gz
--2021-05-03 13:35:43--  https://dl.bintray.com/boostorg/release/1.70.0/source/boost_1_70_0.tar.gz
Resolving dl.bintray.com... 52.32.150.221, 52.43.227.140, 34.213.218.48, ...
Connecting to dl.bintray.com|52.32.150.221|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2021-05-03 13:35:43 ERROR 403: Forbidden.
```

From doing some digging it seems that bintray has rate-limits put on their files.  Other projects have moved away from bintray because of this issue.

foreachthing, 2021-05-03:
Get it from here: https://www.boost.org/users/history/version_1_70_0.html

lukasmatena, 2021-05-03:
I don't know what commit you're building, but the deps scripts download Boost 1.75, not 1.70. And the URL was recently updated with 279113900209ba624c71020fde22ea9ac089e04e. All should be fine, no?

TheAndr0id, 2021-05-03:
This is the 2.3.1 release found here:
https://github.com/prusa3d/PrusaSlicer/archive/refs/tags/version_2.3.1.tar.gz

(got by clicking on the "Source code (tar.gz)" link from the release page for 2.3.1.

The above change has the master tag, but not a 2.3.1 tag.  I'm guessing it didn't get included in the latest release.

Line 16: in the deps-linux.cmake in the 2.3.1 release reads:
`    URL "https://dl.bintray.com/boostorg/release/1.70.0/source/boost_1_70_0.tar.gz"`

lukasmatena, 2021-05-04:
@TheAndr0id 2.3.1 was a minimal patch release of 2.3.0. This change was indeed not included, along with many others that will only appear in 2.4.0.

TheAndr0id, 2021-05-04:
That is good to know, but this also impacts the 2.3.x releases.  I'm a user, not a developer so ensuring production releases work is just as important as fixing issues in the master branch. 

If the changes need to be vetted for the current release then vet them.  Someone published the 2.3.1 release without checking if the build actually worked.  Missing/blocked download URLs is not a tested release.

I'm just sayin...

lukasmatena, 2021-05-04:
> I'm a user, not a developer

If you were a user, this issue would not affect you at all. You would download an AppImage, would not care about how PrusaSlicer is built and would not educate us about untested releases. Which is IMHO quite off. The 2.3.1 release of PrusaSlicer (!!!) was well tested and surprisingly, it could also be built. The whole deps directory only contains utility scripts to make the build easier and PrusaSlicer can be built without them. They most likely even worked at the time of the release. Anyway, like I said, it is fixed on master, the fix will make it into either 2.4.0 or maybe a 2.3.2.

TheAndr0id, 2021-05-05:
Now I'm starting to get angry.

I reported a problem, fix it or not.  If you don't want to bother dealing with your past releases that's your issue.

I won't be submitting any further issues.  I don't need the headaches. 

You can close the ticket, fix it, leave it open, fire it into the sun.  I don't care any more.

lukasmatena, 2021-05-05:
@TheAndr0id We are thankful for bug reports. But writing "Someone published the 2.3.1 release without checking [...]", "[it is]not a tested release" is not reporting issues. I see it as an accuse of sloppiness and because I thought it was quite undeserved in this case and not even true, I responded the way I did. If that's not how you meant it, I'm sorry.

> If you don't want to bother dealing with your past releases

Past releases cannot be amended, if you're suggesting to add this fix into 2.3.1, I don't see how to do it.

Fixed on master with 2791139, closing.

closed 2021-05-05 by lukasmatena