prusa3d/PrusaSlicer #6482: Object sliced incorrectly

mdjurfeldt, 2021-05-04:
### Version
version_2.4.0-alpha0-254-g00835c736

### Operating system type + version
Debian Bullseye

### Behavior
The following project is sliced incorrectly:

[base grid 3x4 square.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/6422163/base.grid.3x4.square.3mf.gz)

I get solid infill at 1mm height inside the squares. (I of course expect the squares to be open and empty.)

rtyr, 2021-05-04:
I was not able to reproduce it with provided 3MF, but I think it will be related to some errors in the model.

Does it happen with STL attached below?
[base grid 3x4 square_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/6423430/base.grid.3x4.square_fixed.zip)

mdjurfeldt, 2021-05-04:
Yes, I provide an example screenshot 
[image]
where it happens. It doesn't seem to be sensitive to any setting I've found yet. I always get the infill, also for your corrected STL.

Did you slice using PrusaSlicer commit 0835c736?

rtyr, 2021-05-04:
You are right, I was able to reproduce it on current master.

bubnikv, 2021-05-05:
Fixed with 7d4b3f2
Thanks for heads up.

closed 2021-05-05 by bubnikv