prusa3d/PrusaSlicer #6485: Wipe tower doesn't apear for MMU2S  interface/support

davidkerny, 2021-05-05:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

davidkerny, 2021-05-05:
It used to be possible to print interface layers with a different material using MMU2s in PrusaSlicer. 
No longer an option, using a different material for supports/interface disables the wipe tower. Why?
The window in screenshot 2 shows what appears when trying to print interface/support with different material. 
It mentions soluble support for some reason, even though this includes any type of material, not only soluble ones.
The variables are named as slugs, not in the same way those appear  to the user in prusaslicer (support_material_extruder vs. Support material/raft/skirt extruder)

I remember using this function together with the wipe tower previously. 
If I run the resulting gcode in a printer, does it mean I have to pull one and push another filament into a single position in MMU2S on each filament change for supports/object?

[image]
[image]

rtyr, 2021-05-05:
Change contact Z distance to 0 or just use bundled `0.20mm SOLUBLE INTERFACE` profile.

bubnikv, 2023-02-24:
@davidkerny What do you recommend? That slicer tells the user to set the support contact gap Z to zero if a soluble interface is selected?

closed 2026-09-29 by prusaslicer-bot[bot]