prusa3d/PrusaSlicer #6486: PrusaSlicer quits unexpectedly while saving a color/swatch in the color picker dialog.

Slagmeulen, 2021-05-05:
### Version
2.3.1+arm64

### Operating system type + version
Mac OS 11.3.1

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S+ MMU2S 
Firmware 3.9.3

### Behavior
PrusaSlicer crashes/closes "PrusaSlicer quit unexpectedly."

1. Open Prusa Slicer;
2. Change Filament Settings (By using the Tab Page or the gear button behind the filament selector);
3. Click on the Color opening the Colors dialog;
4. Drag (holding the left mouse button) the current color to the the first color save location (20 squares/swatches next to the color picker icon);
5. Release the color on the first save square/swatch.

Expected result: color is saved. 
Actual result: color is not saved, PrusaSlicer craches.

#### Project File (.3MF) where problem occurs
No file needed, empty project.

#### Details from the crash log:
Application Specific Information:
Performing @selector(controlAction:) from sender wxNSButton 0x10f6c2380
Assertion failed: (backingScaleFactor > 0.f), function drawnFocusRingBitmapForTime, file /System/Volumes/Data/SWE/macOS/BuildRoots/2288acc43c/Library/Caches/com.apple.xbs/Sources/AppKit/AppKit-2022.44.151/AppKit.subproj/NSAutomaticFocusRing.m, line 306.

notmatt, 2021-12-08:
Just ran into this on macos 11.6, running both 2.4.0b2 and b3; didn't dig up the crash logs since I'm assuming they're similar to the above, but happy to provide any extra information if it's helpful.

bubnikv, 2021-12-08:
Our @FidelCapo cannot reproduce the issue. We would certainly be thankful for the extra information to reproduce the issue. Maybe a video would help us? I don't know.

closed 2026-09-29 by prusaslicer-bot[bot]