prusa3d/PrusaSlicer #6491: Booby Trap: Orange Ball Missing in Action

jackthese, 2021-05-06:
PrusaSlicer 2.3.1 and previous
Windows 10 64 bit
MK3S
I've been perplexed and haunted for over a year by an issue where the Orange Ball and + sign randomly disappeared when I tried to select a layer for a filament color change. I've worked this issue over in the Prusa Forum for nearly a year before user neophyl mentioned something in passing that turned out to cause this behavior: If the Print Settings>>Output options>>Sequential printing>>Complete individual objects" is checked, filament changes are not possible, apparently. Seems like this could be easily implemented, but short of that, an error message or informational message saying why filament changes are suppressed should be added. All this might be terribly obvious to sophisticated users, including the development team, but for us civilians it was a puzzle. Thanks...Ripper

lukasmatena, 2021-05-06:
Closing as a duplicate of #5437. I personally agree that it would not hurt to have the button always visible.

closed 2021-05-06 by lukasmatena

jackthese, 2021-05-06:
Well, I don't think this issue is closed. A fundamental tenet of good user interface design is to put critical information where it's needed, and PrusaSlicer 2.3.1 fails miserably in this case. Read the forum, and you'll see that many people have wasted valuable time because of this poorly flagged issue. Remember, you're designing this program for the average user, not yourself. There should be a warning pop-up when a user clicks or right-clicks the layer selector that says "Filament changes are not available because "Complete individual objects" is selected in the printer tab>>Output options." The program code already knows not to display the ball--all that is needed is for it to display the pop-up.

bubnikv, 2021-05-06:
We agree with you 110%. We just need to squeeze it into our busy
development schedule.