prusa3d/PrusaSlicer #6493: Speicherfehler Slicer V 2.3.0 32 bit

willi239, 2021-05-06:
Hallo
Beim slicen von etwas größer Dateien und Infill drößer 30%, bricht der Slicer mit der Fehlermeldung - warscheinlich Speichergröße reicht nicht - ab. Das 3D Modell passt aber bequem in den Druckraum. Drucker Prusa MK3S+

rtyr, 2021-05-06:
Google translate:

> Hello
> When slicing slightly larger files and infill more than 30%, the slicer breaks with the error message - memory size is probably insufficient. The 3D model fits comfortably in the printing room. Prusa MK3S + printer
> 

Please provide your 3mf project file, PrusaSlicer version, operating system.

willi239, 2021-05-06:
Hallo
projekdatei im Anhang. Prusa Slicer V 2.3.0 Win 10 32 bit. Infill 40%

(#6493)

rtyr, 2021-06-07:
No response. 32-bit system can address only limited amount of memory. Closing.

closed 2021-06-07 by rtyr