prusa3d/PrusaSlicer #6495: Flatten the base of the model

fjpanag, 2021-05-06:
### Version
Version 2.3.0+linux-x64-GTK3

### Operating system type + version
Fedora 33

### 3D printer brand / version + firmware version (if known)
Custom delta based on Duet3D.

### Behavior
Unfortunately, many models do not have a "flat" base.  
The first layer only gets part of the base of the model, with the rest of if being printed on layer 2.  
When this happen, the second layer is not properly squished on the build plate, and I get adhesion problems.

See the attached project as an example.  
(Although I have seen plenty of models with this issue, it is very common.)

I have tried to adjust the settings to fit the whole base of the model on the first layer (like making the first layer very thick), but it does not solve the issue.

I would propose a new feature in Prusa Slicer, that allows this to be fixed easily.  
I can think of two different approaches:

Either clip the model at the specified height.  
A section of the model is made with a plane parallel to the build plate. Only the upper part is kept.  
Essentially everything bellow is discarded, keeping an even surface.

Or extend every perimeter of the model up to a specified height, to the build plate.  
This will "fill" all unevenness of the model, again producing a flat base.

A warning of this condition would be nice to have.  
Prusa slicer may check for unsupported perimeters on layer 2, and warn of potential uneven base of the model.

_Is this a new feature request?_
Yes.

#### Project File (.3MF) where problem occurs
[base.zip](https://github.com/prusa3d/PrusaSlicer/files/6437927/base.zip)

lukasmatena, 2021-05-07:
> A section of the model is made with a plane parallel to the build plate. Only the upper part is kept.

You're describing a 'Cut' function, which is accessible from the left panel and has been there for ages :-)

neophyl, 2021-05-07:
Erm you do know theres a cut tool right ?  That allows you to trim off parts of the model and is used in exactly these cases for this reason.  You can choose the height of the cut and which part to keep, or keep both.
https://help.prusa3d.com/en/article/cut-tool_1779  

Or you could enable rafts and supports to support your lower surface if you cant cut it flast for some reason.

fjpanag, 2021-05-08:
Oh...! I am totally blind!  
I never saw this, which is exactly what I need.

May I keep the issue open for the "extend" functionality?  
If it's a "won't do" the developers may close it.