prusa3d/PrusaSlicer #6496: Skirt interferes with model

fjpanag, 2021-05-06:
### Version
Version 2.3.0+linux-x64-GTK3

### Operating system type + version
Fedora 33

### 3D printer brand / version + firmware version (if known)
Custom delta based on Duet3D.

### Behavior
The "skirt" is usually used for priming the extruder before a print.  
It is however very useful to quickly visualize the size of the object to be printed, get a better idea instead of guessing through your computer screen.

Please have a look at the attached model.  
There are two issues with this:
1. The skirt interferes with the model. When layer 2 is printed, it will be fused to the skirt.
2. You cannot get an idea of the actual size of the printed object.

(Regarding 1. I know that the model is broken. I just use it as an example. There are other cases though that the model has a great overhang, and a similar issue is created.)

I believe that it would be much better to have the skirt follow the extremities of the model, instead of only layer 1.  
It will be a good work-around for problems like 1. and it will greatly help visualizing the print as in 2.  
Or maybe it can be configurable as to which perimeter to follow.

_Is this a new feature request?_
Hmm, yes I think...

#### Project File (.3MF) where problem occurs
[base.zip](https://github.com/prusa3d/PrusaSlicer/files/6437952/base.zip)

lukasmatena, 2021-05-07:
I did not try to print it, but I think that the first layers will have more problems than just the skirt. You'll probably still need to cut off the base of the skull, which will also fix this.

> I believe that it would be much better to have the skirt follow the extremities of the model, instead of only layer 1.

The shape of the skirt is determined by the height of the skirt. If the skirt is 10 mm tall, it takes the largest of all slices up to that height and offsets it outward (depending on 'skirt_distance'). It would in principle be possible to use one more layer for the calculation, but I have an impression that the issue only affects models that are unprintable anyway. And I also don't remember anyone reporting this in the past.

fjpanag, 2021-05-08:
I just can't find an good example now, I cannot remember on which model I had this actual issue.  
But it has happened to me, to have a perfectly printable model, that I did actually print correctly, but I had the skirt fused to it, due to some severe overhand on the first layers.

I would definitely prefer to use the whole model for the skirt calculation, but nevertheless, having it look "up" a couple of layers would be indeed beneficial.

drakonstudio, 2021-05-17:
Here is a simple, perfect example of this [issue.]([Peace_key_chain.zip](https://github.com/prusa3d/PrusaSlicer/files/6491659/Peace_key_chain.zip))

jasonehines, 2021-08-21:
I'm experiencing this with 2.3.3 Windows 10 and the Creepy_Hand.stl in this set [Reaching Hands](https://www.thingiverse.com/thing:3923735). When scaling the model it is fine at 66%, at 67% you can see the skirt beginning to have a straight line on the back side, at 70% it's under the model.

jasonehines, 2021-08-21:
> I'm experiencing this with 2.3.3 Windows 10 and the Creepy_Hand.stl in this set [Reaching Hands](https://www.thingiverse.com/thing:3923735). When scaling the model it is fine at 66%, at 67% you can see the skirt beginning to have a straight line on the back side, at 70% it's under the model.

upon further research, I found that the initial model was imported at a very slight angle (impossible to see in the preview). When I used the "Place on face" feature, it no longer has the skirt issue.

(instead of deleting my comment I thought it would be good to leave with this explanation in case it helps someone else)

lukasmatena, 2024-02-18:
Closing, the model does not have a flat base and it is printed that way.

closed 2024-02-18 by lukasmatena