prusa3d/PrusaSlicer #6500: Slicer profiles 1.2.4 show USD pricing instead of EUR pricing after updating

broodro0ster, 2021-05-07:
### Version
2.3.1

### Operating system type + version
Issue occurs on MacOS and Windows

### 3D printer brand / version + firmware version (if known)
MK3s

### Behavior
After updating the Prusa slicer profiles to 1.2.4, the cost price per spool isn't correct anymore. 
I used to have the EUR values of the spools (for example 24.99 for Prusament PLA)and right now I'm seeing values that matches the conversion rate in USD. The Prusament PLA cost now shows up as 30.24 while it used to be 24.99. The same is true for the other filament profiles.
I live in Europe and I would like to see the EUR value of the spools like it was before the update.

#### Project File (.3MF) where problem occurs
Not needed here. Just go to the filaments tab and check the cost for the filaments.

rtyr, 2021-05-08:
It is currently not possible to store values for more currencies, we chose USD as a default. So it's intended.

It would make sense to make the money/kg field editable, even for system profiles.

broodro0ster, 2021-05-08:
Thanks. I didn't know USD values are now the intended value. I was a bit confused by this change since it has been always EUR for the past years. And if change my slicer language to dutch, it even shows 30.24**€/**kg.

Being able to update the price of the system profiles would be nice. I really like the default Prusa profiles and it would make sense to save them as a duplicate just for the price adjustment.
I hope this can be fixed in a future update :-)

rtyr, 2021-05-08:
Bundled filament costs were always meant to be in USD, but there were some old inaccuracies (VAT not included, cost/kg vs 750g spools, etc.), which were fixed in the last configuration bundle.

>  it even shows 30.24**€/**kg

That is incorrect translation of `money/kg`, we should fix that.