prusa3d/PrusaSlicer #6502: BUG in G-CODE: Missing retraction in some movements

zbig63, 2021-05-09:
PrusaSlicer Version
2.3.0  

Operating system type + version
Windows10  64bit

3D printer brand / version + firmware version (if known)
Custom build  (Repetier GCode flavour)

Behavior

In resulting G-code file some movements were made without previous retraction, even when move length is longer than "minimum travel after retraction" parameter (2mm in my case).
This bug causes serious problems in prints: when nozzle is moved over already printed part without retraction, some amount of filament leaks (or rather is sucked out) from nozzle and glued into printed part, so when next perimeter starts to print there is missing filament in nozzle and beginning of perimeter is simply missing. In my print it causes visible gap in perimeter.  
[Hermes_prod_rec.zip](https://github.com/prusa3d/PrusaSlicer/files/6447025/Hermes_prod_rec.zip)

Bad code looks like that:

G1	X85.779	Y107.154	E1.80653
G1	X85.919	Y95.819	F4800.000
;TYPE:External	perimeter	 	 

Second instruction (movement without extration) is 11,3mm  so it is long enough and should be made with retraction.
This bug is repeated many times in a file. For example such bad movements are as follow:
line    code movement                     len 
740	G1	X85.919	Y95.819	F4800.000	11,3mm
788	G1	X107.030	Y108.500	F4800.000	2,2mm
791	G1	X85.600	Y106.830	F4800.000	2,2mm

I noticed that in each bad movement without retraction one axis offset is shorter than "minimum travel" parameter, where other is longer than this parameter or distance (diagonal) is longer. maybe bug is related to distance calculation?  

Btw. In other part of the file there is proper retraction (1.5mm) before movement like this (retraction is in second line):

G1	X80.607	Y89.951	E12.26002
G1	E10.76002	F2100.000	 
G92	E0	 	 
G1	X88.792	Y97.063	F4800.000
G1	E1.54000	F2100.000	 
;TYPE:External	perimeter	 	 
G1	F1800.000	 	 
G1	X89.267	Y96.637	E1.59801

rtyr, 2021-05-09:
Disable the `Only retract when crossing perimeters` option (Print settings -> Infill).

[image]

Closing.

closed 2021-05-09 by rtyr

zbig63, 2021-05-09:
Oooops,  I do not noticed this option. My fault. With this option unchecked all movements are with retraction now.

zbig63, 2021-05-09:
Btw. moving along existing perimeter should not be treated this same as as crossing perimeter ?